platformio/platformio-core #528: Refactor platform build scripts

ivankravets, 2016-02-21:
Need to refactor all build scripts and put base flags to appropriate SCOPES. More details http://docs.platformio.org/en/latest/projectconf.html#build-flags

User should have ability to pass own build flags and override base settings. In this case, user's flags should follow after "base flags".

ivankravets, 2016-02-21:
Related issue https://github.com/platformio/platformio-atom-ide/issues/54 where `build_flags = -Wno-comment` doesn't work.

closed 2016-02-22 by valeros

sstaub, 2016-03-01:
I tried to use build_flags = -Wno-comment in platformio.ini, but it doesn't work.

ivankravets, 2016-03-01:
@sstaub do you use the latest development version? `platformio --version`

sstaub, 2016-03-01:
The IDE is v1.04

sstaub, 2016-03-01:
Sorry, version of platformio is 2.7.1

> Am 01.03.2016 um 10:25 schrieb Ivan Kravets <notifications@github.com>:
> 
> platformio --version

ivankravets, 2016-03-01:
@sstaub Gear Icon(Settings) on the PlatformIO Toolbar > Packages > platformio-ide > Settings > Use development version of PlatformIO

P.S: Don't forget to uncheck it when 2.8.5 will be released.

sstaub, 2016-03-01:
It did the check in the settings long times ago. But i don’t know which version is now currently installed:

ivankravets, 2016-03-01:
Then Menu > PlatformIO > Upgrade PlatformIO.

It will install the latest development version.

sstaub, 2016-03-01:
The problem is that I do upgrading (see screenshot), but when I ask which version installed 
with platformio —version it says 2.7.1

ivankravets, 2016-03-01:
Open system Terminal (not in IDE), run `pip uninstall platformio`. Restart IDE

sstaub, 2016-03-01:
No changes.

ivankravets, 2016-03-01:
Which is your OS?

sstaub, 2016-03-01:
MacBookPro Yosemite 10.10.5

ivankravets, 2016-03-01:
Give me please an output:
1. Open terminal in IDE: `which platformio`
2. Open System OS terminal: `which platformio`

sstaub, 2016-03-01:
> 
> Give me please an output:
> 1. Open terminal in IDE: which platformio
stefans-mbp-3:OSCwing sstaub$ stefans-mbp-3:~ sstaub$ which platformio
bash: stefans-mbp-3:~: command not found
stefans-mbp-3:OSCwing sstaub$ /usr/local/bin/platformio
Usage: platformio [OPTIONS] COMMAND [ARGS]...

> 2. Open System OS terminal: which platformio
stefans-mbp-3:~ sstaub$ which platformio
/usr/local/bin/platformio
stefans-mbp-3:~ sstaub$

ivankravets, 2016-03-01:
Hm... And `pip list` in the both Terminals (IDE + OS)

sstaub, 2016-03-01:
says platformio (2.7.1) in both terminals

ivankravets, 2016-03-01:
Please mail me@ikravets.com directly.

commit 17796de6 (2016-02-22): Refactor platform build scripts // Resolve #528