platformio/platformio-core #529: Programming using C language only

baronleonardo, 2016-02-21:
I have a board of type "lptm4c1230c3pm",
and I want to program in pure C language without using the style of Energia.

ivankravets, 2016-02-21:
See native example https://github.com/platformio/platformio/tree/develop/examples/titiva/titiva-native-blink

The `framework = energia` here is used for driverlib and LD script. The high level code of Energia's framework will not be used.

baronleonardo, 2016-02-21:
what I want is to write my own main with my own port(s) configuration
I wrote an example
[Sample](http://pastebin.com/cE4ZwGtc)

ivankravets, 2016-02-21:
@valeros please help @baronleonardo

valeros, 2016-02-22:
Hi @baronleonardo 
Could you try such a workaround (latest file versions from TI website are used):
1. Put new [linker file](https://gist.github.com/valeros/951f6cab58a2e9dee5b1) beside your `platformio.ini`.
2. Add new [startup file](https://gist.github.com/valeros/6bddb052985a120bf148) to `src` directory.
3. Add new [library](https://dl.dropboxusercontent.com/u/10025728/stdlib.zip) in `lib` directory (needed by linker script).
4. Alter your `platformio.ini` according to this example:

_platformio.ini_:
```
[env:lptm4c1230c3pm]
platform = titiva
board = lptm4c1230c3pm
build_flags = -Wl,-Tproject.ld -Wl,--entry=ResetISR
```

baronleonardo, 2016-02-27:
There is a strange problem!
I can't get any delay function work!!

this is a sample of a delay function
```
void delay(unsigned long halfsecs)
{
  unsigned long count;
  while(halfsecs > 0 ) { // repeat while still halfsecs to delay
    count = 1538460;
// originally count was 400000, which took 0.13 sec to complete
// later we change it to 400000*0.5/0.13=1538460 that it takes 0.5 sec
    while (count > 0) {
      count--;
    } // This while loop takes approximately 3 cycles
    halfsecs--;
  }
}
```

baronleonardo, 2016-03-04:
@ivankravets

valeros, 2016-03-04:
@baronleonardo Could you provide more details about your problem, please?

baronleonardo, 2016-03-04:
I tried to make a simple delay function using something like the above example
all my code runs as it should be .. but without delay!
like the delay function doesn't do its job at all ..
on the contrary, if I use the default project template ( not this customized one to program using only C )
it works perfectly
I think there is a problem with `startup_gcc.c` .. may be something missing or something wrong IDK

valeros, 2016-03-04:
What is your MCU working frequency?

baronleonardo, 2016-03-04:
my microship is Tiva TM4C123GH6PM, and I think it works with 80MHz clock

valeros, 2016-03-04:
Hmm, let's calculate:
```
count value * 3 cycles = 1538460*3 = 4615380 cycles
1 cycle = 1/F_CPU = 1/80MHz = 12.5 ns
delay = 4615380 cycles * 12.5 ns = ~0.058s
```
Am I missing something?

baronleonardo, 2016-03-04:
I have another delay I'm working with now
I tried `msec` inputs that vary from 1 to 2 millions
still no delay!
and why it works perfectly on the default template!
```
void Delay1ms(unsigned long msec)
{
// write this function
	unsigned long counter;

	while( msec > 0 )
	{
		counter = 16000;
		while(counter-- > 0);
		msec--;
	}

}
```

valeros, 2016-03-04:
What kind of "default template" do you mean?

baronleonardo, 2016-03-04:
I mean the default way to create a project for that kind of micro controller
`platformio init --ide eclipse --board lptm4c1230c3pm`

valeros, 2016-03-04:
Looks like linker optimized delay function.
Could you try to disable optimization? Just add next line to `platformio.ini`:
`build_flags = -O0`

baronleonardo, 2016-03-04:
Nah didn't work either

valeros, 2016-03-04:
Try please with this [linker script](https://gist.github.com/valeros/eacb70f6fe507758b7c9).

baronleonardo, 2016-03-04:
I tried the new linker script with linked optimization and without it
both failed :/

valeros, 2016-03-07:
@baronleonardo Sorry, can't help you here. I don't have this board to debug.
At last you can try this delay:
```
void Delay1ms(unsigned long msec) {
    volatile unsigned long counter;
    while( msec > 0 )  {
        counter = 16000;
        while(counter-- > 0);
        msec--;
    }
}
```

baronleonardo, 2016-03-13:
I'm so so sorry
the changes you gave me on [comment] (https://github.com/platformio/platformio/issues/529#issuecomment-187134776) is working perfectly
I think the problem is about the clocking I used
thank you .. you saved me :+1:

ivankravets, 2016-03-13:
@baronleonardo Thanks a lot for the report! Please grant @valeros with star :star: https://github.com/platformio/platformio/stargazers.

baronleonardo, 2016-03-13:
I want to, but how :D

ivankravets, 2016-03-13:
@valeros likes these stars... :blush: Take a look at the top of this page where are buttons `Watch | Stars | Fork`.

baronleonardo, 2016-03-14:
yeah yeah got it

baronleonardo, 2016-03-27:
I need to use `<stdio.h>` to use `malloc` to create a struct in the heap
but it complains from `undefined reference to `_sbrk`

valeros, 2016-03-28:
You need to implement `_sbrk` function or use default implementation in "nosys" library `build_flags = -lnosys` and don't forget to specify `heap` section and `end` symbol in your linker script.

nayabbashasayed, 2016-06-01:
@baronleonardo
Did you take care of that delay issue? I am also facing the same problem. Except that everything works fine.
If yes, please provide a solution.

baronleonardo, 2016-06-02:
it was because of I didn't initialize and configure `PLL`
the delay issue I face, was because I used the old-fashion while loop only without `systick`

try to initialize and configure `PLL` and `systick`

ivankravets, 2016-12-04:
If you have any questions, please create new topic here https://community.platformio.org

closed 2016-12-04 by ivankravets