platformio/platformio-core #530: platformio forum

malachib, 2016-02-22:
Requesting a forum where us end-users can interact without having to always bother YOU Ivan =)

I propose it be broken down by platform & framework

ivankravets, 2016-02-22:
How about to use our https://www.reddit.com/r/platformio/ ? It isn't popular but we can promote it on the our site.

malachib, 2016-02-22:
In this context reddit appears fairly unstructured.  I'd love to be able to go to say an 32u4 or STM32 or mbed dedicated area and see what's being discussed -- and have that distinct from your official announcements, etc.  I am not so sure reddit works well for that

ivankravets, 2016-02-22:
How about https://www.discourse.org ?

ivankravets, 2016-02-22:
Also, who will support it?

malachib, 2016-02-22:
discourse.org looks promising.  I can't offer real / full support, but I can lend a hand moderating here and there

ivankravets, 2016-03-09:
Welcome to PlatformIO Community :) http://community.platformio.org

closed 2016-03-09 by ivankravets

malachib, 2016-03-10:
nice!!

commit a9bd45bb (2016-03-09): Launch PlatformIO Community Forums // Resolve #530