platformio/platformio-core #533: blinky on nRF51DK + mbed framework doesn't work

NAzT, 2016-02-22:
I've tried mbed framework on nucleo_f411re and nucleo_f303k8 the result is works like a charm but 
doesn't work on nrf51.

here's my config. 
[env:nrf51_dk]
platform = nordicnrf51
framework = mbed
board = nrf51_dk
upload_port = /Volumes/JLINK
====

.hex header from mbed online compier  
:020000040000FA
:10000000C0070000D1060000D1000000B1060000CA
:1000100000000000000000000000000000000000E0
:100020000000000000000000000000005107000078
:100030000000000000000000DB000000E500000000
:10004000EF000000F9000000030100000D010000B6

but .hex from platformio is 
:020000021000EC
:10C000000080002025D4010061D4010063D4010028
:10C010000000000000000000000000000000000020
:10C0200000000000000000000000000065D40100D6
:10C03000000000000000000067D4010069D4010086
:10C040006BD401006BD40100E5D501006BD4010075
:10C050006BD40100000000006BD401006BD4010020

valeros, 2016-02-22:
We have the same issue https://github.com/platformio/platformio/issues/500 and working on it.

malachib, 2016-02-22:
+1

might be experiencing the same thing with my Nucleo board also.

NAzT, 2016-02-23:
Thanks, @valeros looking forward to the next version!

commit 9d1128af (2016-08-02): Implement firmware merging with base firmware for Nordic nRF51 development platform // Resolve #500 , Resolve #533
closed 2016-08-02 by ivankravets