platformio/platformio-core #534: Websockets for Arduino ESP8266 throws errors

kristsm, 2016-02-24:
I installed websockets library but even then compiler throws following errors:

scons: *** [.pioenvs\esp12e\WebSockets_ID549\WebSockets.o] Error 1
.pioenvs\esp12e\WebSockets_ID549\WebSocketsClient.cpp: In member function 'void WebSocketsClient::begin(const char*, uint16_t, const char*, const char*)':
.pioenvs\esp12e\WebSockets_ID549\WebSocketsClient.cpp:67:16: error: 'RANDOM_REG32' was not declared in this scope
randomSeed(RANDOM_REG32);
^
scons: *** [.pioenvs\esp12e\WebSockets_ID549\WebSocketsClient.o] Error 1
 [ ERROR ] Took 13.12 seconds 

It seems ESP8266 core files are not up to date. Can I solve this myself like - updating Platformio ESP8266 arduino framework with latest official ESP8266 Arduino github repository?

Thanks!

valeros, 2016-02-25:
Hi @kristsm `RANDOM_REG32` is defined in RC framework version, please follow [these instructions](https://github.com/platformio/platformio/issues/401#issuecomment-166958952) to use the latest framework.

kristsm, 2016-02-25:
Hi, valeros! Thanks for the help! Your instructions healed compiling problem. 

Still Websockets are not working, but it might also be related to my code.

ivankravets, 2016-02-28:
@kristsm please provide project to reproduce this issue

kristsm, 2016-02-29:
Hi! No worries, I checked the websocket part of the code and everything is working fine !

ivankravets, 2016-02-29:
@kristsm please https://twitter.com/platformio_org/status/704304862998224896 and report us.

ivankravets, 2016-02-29:
Please reopen this issue if you still have problems

closed 2016-02-29 by ivankravets