platformio/platformio-core #535: Support for Arduino/Genuino 101

maxbrueckl, 2016-02-24:
It is based on Intel Curie SOC(https://www.arduino.cc/en/Main/ArduinoBoard101). So there is no platformio package for this yet.

Toolchain is ARC GNU Toolchain located at https://github.com/foss-for-synopsys-dwc-arc-processors/toolchain.

Upload with dfu-util(http://dfu-util.sourceforge.net/) and seems to ship with own drivers.

The files are available via Arduino IDE board manager, but the documentation is quite thin until now. Is there any chance, platformio will support this platform? Tell me, if I can help with anything.

ivankravets, 2016-02-25:
Thanks, we will contact with Arduino.cc and Intel Corporation to help us with integration this board to PlatformIO.

ivankravets, 2016-02-27:
@w4ilun, @D3f0 

Do you have any ideas as for this issue?

ivankravets, 2016-02-27:
All info is in https://github.com/platformio/platformio/issues/157#issuecomment-189547354

ivankravets, 2016-04-28:
@maxbrueckl please re-test with the latest http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-04-28 by ivankravets

maxbrueckl, 2016-04-28:
Thank you for your efforts!
I can confirm that it compiles for arduino101 now. 

However, I get an error message, when I try to upload to the board. This is the corresponding console output:

```
E:\Arduino\BalanceBot>platformio run --upload-port COM3 -e genuino101 -t upload
[04/28/16 15:09:58] Processing genuino101 (platform: intel_arc32, board: genuino
101, framework: arduino)
-------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs\genuino101\firmware.bin"])
"C:\Users\max\.platformio\packages\tool-arduino101load\arduino101load" E:\Arduin
o\BalanceBot\.pioenvs\genuino101 .pioenvs\genuino101\firmware.bin "COM3" verbose

Starting download script...
Args to shell: [E:\Arduino\BalanceBot\.pioenvs\genuino101 .pioenvs\genuino101\fi
rmware.bin COM3 verbose]
Serial Port: COM3
BIN FILE .pioenvs\genuino101\firmware.bin
Waiting for device...
exec: "E:/Arduino/BalanceBot/.pioenvs/genuino101/dfu-util": file does not exist
scons: *** [upload] Error 1
========================= [ ERROR ] Took 1.66 seconds =========================
```

valeros, 2016-04-28:
Hi @maxbrueckl !
Could you please test latest PlatformIO and this [uploader](https://www.dropbox.com/s/hqvjsgc749wxb5c/tool-arduino101load.zip?dl=0). Just replace old one in `C:\Users\max\.platformio\packages\tool-arduino101load\`
Thanks!

maxbrueckl, 2016-04-28:
I can confirm, that it all works now. Uploading included. Great work!

Thank you very much.

Notice to everyone working with Arduino101 right now: dfu-util currently does not reset the board before uploading. You need to manually press the "reset" button to bring the board in a flashable state.

maxbrueckl, 2016-04-28:
I know, the issue is solved now, but may I ask a question considering packages in platformio.

As it seems they just exist and get downloaded (concrete toolchain-intelarc32, framework-arduinointel, tool-arduino101load), but I cant find where they are defined in the commits mentioned with this issue. Am I missing something?

If I understand correctly, it would be much more scalable, if they would be treated like boards and platforms (definable by users).

ivankravets, 2016-04-29:
@maxbrueckl the packages are located in @bintray https://bintray.com/platformio

PlatformIO installs package into `~/.platformio/packages` directory.