platformio/platformio-core #536: creation of src/ and lib/ dirs when src_dir and lib_dir are changed

glelouet, 2016-02-25:
My platformio.ini contains

> [platformio]
> src_dir = ./
> lib_dir = ../lib

then when I init (because I want to modify the sources with eclipse), platformio asks me if I want to create the src/ and lib/ dirs. Which I don't but answering no prevents it from creating the eclipse .project.

two options IMO :
 - create the lib_dir and src_dir instead of lib/ and src/
 - add a --no-create option to not check the presence of the dirs

glelouet, 2016-02-25:
PlatformIO, version 2.8.4

ivankravets, 2016-03-07:
I don't handle `lib_dir` because it has other meaning in PlatformIO 2.0.

I hope that we will add `--srd-dir` and `--lib-dir` options to `pio init` command in PlatformIO 3.0.

commit 4063cd6a (2016-03-07): Use src_dir directory from platformio.ini when initializing project otherwise create base src directory // Resolve #536
closed 2016-03-07 by ivankravets