platformio/platformio-core #539: Assembly - lptm4c1230c3pm board

baronleonardo, 2016-02-26:
I want to program pure assembly on lptm4c1230c3pm board
but it gets me lots of error complaining about the .asm file
it acts like it doesn't know what is that!

valeros, 2016-03-01:
@baronleonardo Could you please share your ASM code example?

baronleonardo, 2016-03-01:
```
;=========================================================================================================
; Blink in Assembly
; By Ricardo Angeli (03/04/15)
; Designed for Tiva Connected Launchpad (TM4C1294NCPDT)
;=========================================================================================================
; Description:
; Blinks an onboard LED at an approximate rate of once
; per second.  Designed as a test for an assembly-only
; program on the Launchpad.
;=========================================================================================================

    .global  main                                   ; makes main accessible from outside this file.
    .thumb                                          ; use thumb, duh

    .text                                           ; set memory location to flash
    .align 4                                        ; align on 32-bit boundary

;=========================================================================================================
; Hardware Constants
;=========================================================================================================
GPIO_PORTJ_DATA_R             .field 0x400603FC,32
GPIO_PORTJ_DIR_R              .field 0x40060400,32
GPIO_PORTJ_AFSEL_R            .field 0x40060420,32
GPIO_PORTJ_PUR_R              .field 0x40060510,32
GPIO_PORTJ_DEN_R              .field 0x4006051C,32
GPIO_PORTJ_AMSEL_R            .field 0x40060528,32
GPIO_PORTJ_PCTL_R             .field 0x4006052C,32
PJ0                           .field 0x40060004,32
PJ1                           .field 0x40060008,32
SWITCHES                      .field 0x4006000C,32
SW1_PRESSED                   .equ 0x02              ; value read from location SWITCHES when just SW1 is
                                                     ; pressed.
SW2_PRESSED                   .equ 0x01              ; value read from location SWITCHES when just SW2 is
                                                     ; pressed.
BOTH_PRESSED                  .equ 0x00              ; value read from location SWITCHES when both
                                                     ; switches are pressed
NO_PRESSED                    .equ 0x03              ; value read from location SWITCHES when neither
                                                     ; switch is pressed.
GPIO_PORTN_DATA_R             .field 0x400643FC,32
GPIO_PORTN_DIR_R              .field 0x40064400,32
GPIO_PORTN_AFSEL_R            .field 0x40064420,32
GPIO_PORTN_DEN_R              .field 0x4006451C,32
GPIO_PORTN_AMSEL_R            .field 0x40064528,32
GPIO_PORTN_PCTL_R             .field 0x4006452C,32
PN0                           .field 0x40064004,32
PN1                           .field 0x40064008,32
LEDS                          .field 0x4006400C,32
LED1_ON                       .equ 0x02               ; value written to location PN1 or LEDS to turn on
                                                      ; LED1.
LED2_ON                       .equ 0x01               ; value written to location PN0 or LEDS to turn on
                                                      ; LED2.
SYSCTL_RCGCGPIO_R             .field 0x400FE608,32
SYSCTL_RCGCGPIO_R12           .equ 0x00001000         ; GPIO Port N Run Mode Clock Gating Control
SYSCTL_RCGCGPIO_R8            .equ 0x00000100         ; GPIO Port J Run Mode Clock Gating Control
SYSCTL_PRGPIO_R               .field 0x400FEA08,32
SYSCTL_PRGPIO_R12             .equ 0x00001000         ; GPIO Port N Peripheral Ready
SYSCTL_PRGPIO_R8              .equ 0x00000100         ; GPIO Port J Peripheral Ready

;=========================================================================================================
; Custom Constants
;=========================================================================================================
CPU_FREQ                      .field 8000000,32       ; Current CPU clockspeed stored in Hertz

;=========================================================================================================
; Program Code
;=========================================================================================================
        .align 2                                      ; align on 16-bit boundary

main:                                                 ; must be called main to execute on its own
[... 46 lines cut ...]
        bic r1, r1, #LED2_ON
        str r1, [r0]

        ; Initiate wait drive!
        ldr r2, CPU_FREQ                              ; load number of cycles we want to wait
wait2:
        sub r2, r2, #10                               ; subtract 10 every 10 cycles
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        cmp r2, #0
        bgt wait2                                     ; loop back unless r2 <= 0

        ; loop to the beginning of the blink routine all over again
        b blink

        ; if calling this assembly program from C, this would pop back used registers and return back to C
        pop {r0-r2, lr}
        mov pc,lr

```

valeros, 2016-03-01:
1. Ensure that the assembly code filename has a `.S` suffix.
2. Are you sure this example is written according to GCC assembly syntax? Looks like it's written for TI compiler.

baronleonardo, 2016-03-01:
I tried `.S` suffix
still the same problem
it dob't even understand that `;` means a comment!

valeros, 2016-03-01:
Looks like `;` isn't a valid comment section in GCC asm. Here is [GNU ARM Assembler Quick Reference](http://www.ic.unicamp.br/~celio/mc404-2014/docs/gnu-arm-directives.pdf)

baronleonardo, 2016-03-01:
I'm so sorry .. my mistake
thanks for your help :-)

closed 2016-03-01 by baronleonardo