platformio/platformio-core #540: Installing tool-scons package: - TypeError: coercing to Unicode: need string or buffer, NoneType found

liquiddandruff, 2016-02-27:
Hello,

PlatformIO looks really cool and I can't wait to use it! I was following this [guide](http://www.penninkhof.com/2015/12/1610-over-the-air-esp8266-programming-using-platformio/) but I encountered this error. Wonder what the problem is?

Ubuntu 14.04
PlatformIO 2.8.4

I've been using the arduino command line to upload firmware but couldn't find any info on OTA flashing through the cli, so if this works it'll be so sweet.

```
$ platformio run --target upload
[Fri Feb 26 21:48:35 2016] Processing nodemcu (platform: espressif, board: nodemcu, framework: arduino)
-------------------------------------------------------------------------------------------------------------
The platform 'espressif' has not been installed yet. Would you like to install it now? [y/N]: y
Installing tool-scons package:
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 90, in cli
    results.append(ep.process())
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 126, in process
    result = self._run()
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 192, in _run
    return p.run(build_vars, build_targets, self.verbose_level)
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 426, in run
    self._install_default_packages()
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 412, in _install_default_packages
    return self.install()
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 329, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 111, in install
    assert isfile(dlpath)
  File "/usr/lib/python2.7/genericpath.py", line 29, in isfile
    st = os.stat(path)
TypeError: coercing to Unicode: need string or buffer, NoneType found
```

ivankravets, 2016-02-27:
Try `rm -fr ~/.platformio` and start again. Did it help?

liquiddandruff, 2016-02-27:
I deleted that directory and ran `platformio run --target upload` again, same error.

ivankravets, 2016-02-27:
Strange issue. I don't know how to explain it. Looks like some of dependent package doesn't exist after downloading
Try to install development platform manually:
```
platformio platforms install espressif
```

ivankravets, 2016-02-27:
Do you have firewall enabled in your OS? Try to check if you have access to @bintray via downloading https://dl.bintray.com/platformio/dl-packages/tool-scons-1.tar.gz

liquiddandruff, 2016-02-27:
Hmm. I get the same error if I try to install it manually too.

```
$ platformio platforms install espressif
Installing toolchain-xtensa package:
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/platforms.py", line 40, in platforms_install
    if p.install(with_package, without_package, skip_default_package):
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 329, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 111, in install
    assert isfile(dlpath)
  File "/usr/lib/python2.7/genericpath.py", line 29, in isfile
    st = os.stat(path)
TypeError: coercing to Unicode: need string or buffer, NoneType found
```

I don't have a firewall. I have no problems downloading that file through my browser (chrome) or with wget.

ivankravets, 2016-02-27:
Could you provide me remote session for debug? SSH, TeamViewer? It will save my and your time. Thanks.

ivankravets, 2016-02-27:
OK, let's debug it remotely.

```
pip uninstall platformio
pip install -U https://github.com/platformio/platformio/archive/feature/issue-540.zip
```

Try to `platformio platforms install espressif` and give an output.

liquiddandruff, 2016-02-27:
I'm going to be busy throughout the weekend but I will try the fix as soon as I can. 

Thanks.

ivankravets, 2016-02-29:
@liquiddandruff do you have any news?

Krifto, 2016-03-01:
I had exactly the same problem as liquiddandruff - with your dev version the problem went away, everything installed successfully.

ivankravets, 2016-03-01:
@Krifto But I haven't changed code. I just put a few "prints" to debug.

Could you try?
```
pip uninstall platformio
pip install platformio
```

You should be switched to the latest stable version

Krifto, 2016-03-01:
I did that now, too, and tings are still working. I get the feeling though that the uninstall did not remove the espressif platform.

I am generally confused as to whether I should run these comands with sudo or not (without sudo I get access violations). Apart, I have both python2 and 3 installed, therefore I use pip2 and not the plain pip.

ivankravets, 2016-03-01:
@Krifto if you install platformio with `sudo`, then need to remove it with `sudo` too.

Krifto, 2016-03-01:
Yeah, I figured as much.

Just to confirm again: After uninstalling and reinstalling, I am still able to install platforms successfully.

ivankravets, 2016-03-01:
@liquiddandruff could you repeat the same? 
```
pip uninstall platformio
pip install platformio
```

liquiddandruff, 2016-03-01:
Hi, sorry for the delay.

I ran the following commands after renaming ~/.platformio to something else
```
pip uninstall platformio
pip install -U https://github.com/platformio/platformio/archive/feature/issue-540.zip
platformio platforms install espressif
```

Output

```
$ platformio platforms install espressif

****************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/
- donate to keep PlatformIO alive! > http://platformio.org/#!/donate
****************************************************************************************************************************************************************************************************************************

Installing toolchain-xtensa package:
Bintray is off-line
Downloading  [####################################]  100%          
dlpath /home/steven/.platformio/packages/toolchain-xtensa/toolchain-xtensa-linux_x86_64-2.tar.gz
Unpacking  [####################################]  100%          
Installing tool-esptool package:
Downloading  [####################################]  100%
dlpath /home/steven/.platformio/packages/tool-esptool/tool-esptool-linux_x86_64-7.tar.gz
Unpacking  [####################################]  100%
Installing ldscripts package:
Downloading  [####################################]  100%
dlpath /home/steven/.platformio/packages/ldscripts/ldscripts-19.tar.gz
Unpacking  [####################################]  100%
The platform 'espressif' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
```

So it seems to work now with the dev version, even though as you said only print statements were added. That's weird, since i'm pretty sure I uninstalled and reinstalled platformio a few times but I still got the same error.

Trying again with the stable version now:

```
pip uninstall platformio
mv ~/.platformio ~/._platformio
pip install platformio

```

Output

```
$ platformio platforms install espressif
****************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/
- donate to keep PlatformIO alive! > http://platformio.org/donate/
****************************************************************************************************************************************************************************************************************************

Installing toolchain-xtensa package:
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/platforms.py", line 40, in platforms_install
    if p.install(with_package, without_package, skip_default_package):
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 329, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 111, in install
    assert isfile(dlpath)
  File "/usr/lib/python2.7/genericpath.py", line 29, in isfile
    st = os.stat(path)
TypeError: coercing to Unicode: need string or buffer, NoneType found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

Hmm so it seems I still get this with the stable version?

ivankravets, 2016-03-01:
@liquiddandruff this is unreal "Bintray is off-line". Dev version works because I switched you to our API server which works temporary as DL mirror while we test @bintray. I've just disabled redirection to BinTray. Please try again this and provide here an output:

```
rm -rf ~/.platformio
pip uninstall platformio
pip install -U https://github.com/platformio/platformio/archive/feature/issue-540.zip
platformio platforms install espressif
```

liquiddandruff, 2016-03-01:
Ah I see, that's funny :) Great that the problem is found

I ran those commands, here is the output .

```
$ platformio platforms install espressif

****************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/
- donate to keep PlatformIO alive! > http://platformio.org/#!/donate
****************************************************************************************************************************************************************************************************************************

Installing toolchain-xtensa package:
Bintray is off-line
Downloading  [####################################]  100%          
dlpath /home/steven/.platformio/packages/toolchain-xtensa/toolchain-xtensa-linux_x86_64-2.tar.gz
Unpacking  [####################################]  100%          
Installing tool-esptool package:
Downloading  [####################################]  100%
dlpath /home/steven/.platformio/packages/tool-esptool/tool-esptool-linux_x86_64-7.tar.gz
Unpacking  [####################################]  100%
Installing ldscripts package:
Downloading  [####################################]  100%
dlpath /home/steven/.platformio/packages/ldscripts/ldscripts-19.tar.gz
Unpacking  [####################################]  100%
The platform 'espressif' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
```

ivankravets, 2016-03-01:
I've just added "Debug 2" https://github.com/platformio/platformio/commit/2fbe21066e5fb958381a4e321fe6dfcfdb5a2696

Please repeat commands from https://github.com/platformio/platformio/issues/540#issuecomment-190965979

liquiddandruff, 2016-03-01:
Output. Seems to be an SSL issue.

```
$ platformio platforms install espressif

****************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/
- donate to keep PlatformIO alive! > http://platformio.org/#!/donate
****************************************************************************************************************************************************************************************************************************

Installing toolchain-xtensa package:
Start downloading from  https://dl.bintray.com/platformio/dl-packages/toolchain-xtensa-linux_x86_64-2.tar.gz
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/platforms.py", line 40, in platforms_install
    if p.install(with_package, without_package, skip_default_package):
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 329, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 102, in install
    dlpath = self.download(info['url'], pkg_dir, info['sha1'])
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 44, in download
    fd = FileDownloader(url, dest_dir)
  File "/usr/local/lib/python2.7/dist-packages/platformio/downloader.py", line 45, in __init__
    headers=util.get_request_defheaders())
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 67, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 53, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 468, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 597, in send
    history = [resp for resp in gen] if allow_redirects else []
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 195, in resolve_redirects
    **adapter_kwargs
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 576, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 447, in send
    raise SSLError(e, request=request)
SSLError: [Errno 1] _ssl.c:510: error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

ivankravets, 2016-03-01:
@liquiddandruff Wow! This is very interesting. 

@jbaruch do you have any ideas?

ivankravets, 2016-03-01:
@liquiddandruff please switch to the [latest development version](http://docs.platformio.org/en/latest/installation.html#development-version) where I enabled our backup mirror while we will resolve this issue.

Thanks a lot for the help and debugging.

P.S: As for me, it looks that you have broken root certificate or etc. I'm not guru in security.

ivankravets, 2016-03-01:
@liquiddandruff could you try `sudo update-ca-certificates`?

See more results https://www.google.com.ua/search?client=safari&rls=en&q=SSL3_GET_SERVER_CERTIFICATE:certificate+verify+failed&ie=UTF-8&oe=UTF-8&gfe_rd=cr&ei=MSzWVpe4C8uAYIOHirAI

liquiddandruff, 2016-03-02:
I figured it could be my machine. I ran `sudo update-ca-certificates` with output:

```
Updating certificates in /etc/ssl/certs... 0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d....
done.
done.
```

I ran  `$ platformio platforms install espressif` again but the error is the same

ivankravets, 2016-03-02:
@liquiddandruff Thanks! Let's wait for @jbaruch.

In any case, I will add support for mirroring in the next 2.8.5 release. I'm going to release it this week.

jbaruch, 2016-03-02:
Guys, please **don't** wait for me. Unlike the support I am not available 24/7 :)
The support team will be more than happy to help you, it's support@jfrog.com. 
I am forwarding them this issue now and will make sure that it gets highest priority.

jbaruch, 2016-03-04:
We suspect that it's because the resolver in Platform.io doesn't follow redirects (to CDN). Need to confirm with you, guys.

ivankravets, 2016-03-04:
Guys, sorry for the delayed answer. I've just spent time and found interesting things. 

@jbaruch please sorry for the wrong "ping". This issue isn't no related to @bintray 

----

1. PlatformIO uses [Requests](http://docs.python-requests.org/en/latest/) package for HTTP operations including packages downloading.
2. This package has built-in SSL certificate verification mechanism. It is enable by default. It means, that this package behaves as Web Browser.
3. It has pre-defined location for the different OS where to looking for `*.cert` files. If OS/distributive has specific configuration and package can't find `*.cert`, we will see the error above.
4. We can control the behaviour of this package. What is more, user is able to tell `Requests` own location with certificates. See http://docs.python-requests.org/en/latest/user/advanced/#ssl-cert-verification

---

> You can pass verify the path to a CA_BUNDLE file or directory with certificates of trusted CAs:

@liquiddandruff could try to play with `REQUESTS_CA_BUNDLE` environment variable? For example,

```
 # be sure that development version is deleted, where mirrors are enabled
pip uninstall platformio
# install the latest stable version where mirrors are turned off and Bintray will be used
pip install platformio
# remove previous installed packages
rm -rf ~/.platformio

export REQUESTS_CA_BUNDLE=/path/to/dir/with/certificates
platformio platforms install epsressif
```

liquiddandruff, 2016-03-06:
Hello @ivankravets 

I've recently switched to arch linux and I can't seem to replicate this issue anymore. I installed the espressif platform as usual and everything's fine and dandy. 

I wonder if this problem has something to do with the CA certs when I originally installed 14.04 (I haven't upgraded its packages for a while).

ivankravets, 2016-03-06:
@liquiddandruff please reopen this issue if you still have problems

closed 2016-03-06 by ivankravets