platformio/platformio-core #543: Add an include file to IDE=Visual Studio to fix Intellisense errors

gerritv, 2016-02-27:
When using VS as an IDE and compiling gcc/g++ projects such as for ESP8266, VS Intellisense puts out errors. These are all related to differences between VS idea of c/c++ and those of gcc/Clang. Eg. __asm__, __attribute__ etc. The actual compile works fine.
I propose enhancing the visualstudio .tpl file to include NMakeForceInclude node in the generated .vxcproj file. This item should point to an include file that is part of the platformio package somewhere.

Example .vxcproj change in visualstudio.tpl: (also needs a join construct in case the item already has data.)
<NMakeForcedIncludes>%someconstantdirectoryinplatformiodirectory/IntellisenseClang.h</NMakeForcedIncludes>

Content of IntellisenseClang.h file:
#pragma once
#ifdef _MSC_VER
#define __asm__(x)
#define __extension__(x)
#define __attribute__(x)
#define __builtin_va_list int
#define __extension__
#define __inline__
#define __builtin_constant_p
#define _Bool bool
#undef __cplusplus
#endif

The include file is benign if not being used by Intellisense compiler, that is the only time that _MSC_VER is defined.

Locally tested with ESP8266 and AVR compiles on VS2015CE.

See also ssion in gitchat from 27 Feb

gerritv, 2016-02-28:
After more work on my actual project, I come to conclusion that the IntellisenseClang.h file must be in the local project directory. This is because depending on the type of src file being compiled, the `#undef _cplusplus` should exist or not (exist for .c, program, not exist for .cpp/ino program).

ivankravets, 2016-03-07:
See http://docs.platformio.org/en/latest/ide/visualstudio.html#intellisense-errors

P.S: We can find here comfortable solution for all our users. We will contact with @microsoft to help us resolve this issue in the better way.

closed 2016-03-07 by ivankravets

gerritv, 2016-03-07:
Good solution.
can you please also add something such as:
To set the parameter for a project, go to Project/Properties/NMake and use the ... at rh side of ForcedIncludes to set your Intellisense fix file.
[image]

ivankravets, 2016-03-07:
@gerritv I've just edited your original message https://github.com/platformio/platformio/issues/543#issue-136976192.

Please edit your comment above with described explanation what need to do. Also, you can attach code of `IntelliSenseClang.h` here. We have a link to this issue from docs.

EddieGreen, 2016-07-28:
I'm running VS2015 pro, with Arduino libraries installed for ESP8266

For me I get one intellisense at the top of the code saying
[image]

The addition of the header file mentioned here had no effect.

After digging around I added the following to the project include search path:
`C:\Users\XXXXXXXXXXXXXXX\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2;C:\Users\XXXXXXXXXXXXXXX\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf;C:\Users\XXXXXXXXXXXXXXX\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include;C:\Users\XXXXXXXXXXXXXXX\AppData\Local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\lib\gcc\xtensa-lx106-elf\4.8.2\include;C:\Users\XXXXXXXXXXXXXXX\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\include;%(AdditionalIncludeDirectories);`

This restores intellisense. I added this to 'platformio.vcxproj.tpl'
[image]

Now, would one of you bright chaps explain why this works - I'm just banging keys here...
[image]

While I'm about it, VS complains about the `<PlatformToolset>v120</PlatformToolset>` entry here. 
[image]

Removing it seemed to make sense to me. What do you think?

commit efd18968 (2016-03-07): Explain about IntelliSense Errors for VS // Resolve #543