platformio/platformio-core #545: Infinite loop when running `platformio ci`

loopj, 2016-02-28:
I'm not sure exactly when this started to happen, but in my project there is now an infinite loop when I run `platformio ci`, here's the repo:
https://github.com/loopj/i2c-sensor-hal

I'm using [symlinks inside my example apps](https://github.com/loopj/i2c-sensor-hal/tree/master/examples/arduino/lib) to include to the latest version of my library in `lib`, which seems to be the cause, but I've always done this since I release my library and the infinite loop has only just started to happen within the past few weeks.

Any ideas @ivankravets?

ivankravets, 2016-02-28:
Please be careful with http://docs.platformio.org/en/latest/userguide/cmd_ci.html

1. `platformio ci` accepts SOURCE code as arguments where you specify PROJECT folder. 
2. Try to clone your repo on the UNIX machine. `i2c-sensor-hal-master/examples/arduino/lib/i2c-sensor-hal` is SYMBOLIC link. Let's read docs:

> In a nutshell, using SRC and platformio ci --lib contents PlatformIO initialises via platformio init new project in platformio ci --build-dir with the build environments (using platformio ci --board or platformio ci --project-conf) and processes them via platformio run command.

In this case, you ask PlatformIO to use `i2c-sensor-hal-master/examples/arduino` as SOURCE folder (where it is project). PlatformIO creates temporary folder and tries to copy `i2c-sensor-hal-master/examples/arduino` to `tmp/src`. As I've mentioned above,  `i2c-sensor-hal-master/examples/arduino/lib/i2c-sensor-hal` is SYMBOLIC link and you receive "infinite loop". 

I can't help you in this situation because I ask Python to cope one folder to another. I don't check "what user asks me to copy". See https://github.com/platformio/platformio/blob/develop/platformio/commands/ci.py#L134

---------

Let's discuss your project. You have 100% PlatformIO-based projects within examples. In this case, you don't need to use `pio ci`. Use `platformio run` instead as you do it on your machine.

Foe example `.travis.yml`,
```yml
language: python
python:
    - "2.7"

cache:
    directories:
        - "~/.platformio"

env:
    - PLATFORMIO_PROJECT_DIR=examples/arduino
    - PLATFORMIO_PROJECT_DIR=examples/mbed

install:
    - pip install -U platformio

script:
    - platformio run -d $PLATFORMIO_PROJECT_DIR
```

P.S: If you want to use `pio ci` command then change `- PLATFORMIO_CI_SRC=examples/arduino` to `- PLATFORMIO_CI_SRC=examples/arduino/src`.

P.S.S: Should I add this example with direct `pio run` to http://docs.platformio.org/en/latest/ci/travis.html ?

ivankravets, 2016-02-29:
Please reopen issue if you still need help or have problems.

closed 2016-02-29 by ivankravets

loopj, 2016-03-01:
Thanks Ivan! I've moved to using the simpler travis.yml you recommended and everything now works. I definitely think adding this to the documentation would help