platformio/platformio-core #546: ESP8266 native fails to run on Adafruit Huzzah

NoumanSaleem, 2016-02-29:
Hi all, 

Firstly, thanks for this great project!

I am trying to use platformio to program my Adafruit Huzzah using the native Espressif ESP8266 SDK. Build and upload succeed, but the serial output is covered in errors.

The only changes I've made to the esp8266-native project is modifying the platformio.ini 
```ini
[env:huzzah]
platform = espressif
board = huzzah
```

build:
```bash
Executing: platformio run 
[Sun Feb 28 17:05:09 2016] Processing huzzah (platform: espressif, board: huzzah)
--------------------------------------------------------------------------------
xtensa-lx106-elf-gcc -o .pioenvs/huzzah/src/user_main.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DPLATFORMIO=020804 -I/Users/nouman/.platformio/packages/sdk-esp8266/include -Isrc src/user_main.c
xtensa-lx106-elf-gcc -o .pioenvs/huzzah/firmware.elf -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,-T"/Users/nouman/.platformio/packages/sdk-esp8266/ld/eagle.app.v6.ld" .pioenvs/huzzah/src/user_main.o -L/Users/nouman/.platformio/packages/ldscripts -L/Users/nouman/.platformio/packages/sdk-esp8266/lib -L.pioenvs/huzzah -Wl,--start-group -lc -lgcc -lphy -lpp -lnet80211 -llwip -lwpa -lmain -ljson -lupgrade -lsmartconfig -lpwm -lat -lssl -Wl,--end-group
"xtensa-lx106-elf-size" -B -d .pioenvs/huzzah/firmware.elf
"/Users/nouman/.platformio/packages/tool-esptool/esptool" -eo .pioenvs/huzzah/firmware.elf -bo .pioenvs/huzzah/firmware_00000.bin -bm qio -bf 40 -bz 512K -bs .text -bs .data -bs .rodata -bc -ec -eo .pioenvs/huzzah/firmware.elf -es .irom0.text .pioenvs/huzzah/firmware_40000.bin -ec -v
text	   data	    bss	    dec	    hex	filename
217973	   1356	  37808	 257137	  3ec71	.pioenvs/huzzah/firmware.elf
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
using ELF file ".pioenvs/huzzah/firmware.elf"
created structure for binimage ".pioenvs/huzzah/firmware_00000.bin" with entry address 0x40100004
setting flash mode from qio to qio
setting flash frequency from 40 to 40
setting flash size from 512K to 512K
added segment #0 to binimage for address 0x40100000 with size 0x0000626C
added section .text at 0x40100000 size 0x0000626A with padding 0x00000002
set bimage entry to 0x40100004
added segment #1 to binimage for address 0x3FFE8000 with size 0x0000054C
added section .data at 0x3FFE8000 size 0x0000054C
set bimage entry to 0x40100004
added segment #2 to binimage for address 0x3FFE8550 with size 0x00000544
added section .rodata at 0x3FFE8550 size 0x00000544
set bimage entry to 0x40100004
saved binimage file, total size is 27936 bytes, checksum byte is 0x4B
using ELF file ".pioenvs/huzzah/firmware.elf"
saved section ".irom0.text" to file ".pioenvs/huzzah/firmware_40000.bin"
========================= [SUCCESS] Took 1.00 seconds =========================
```

upload:
```bash
Executing: platformio run --target upload 
[Sun Feb 28 17:05:43 2016] Processing huzzah (platform: espressif, board: huzzah)
--------------------------------------------------------------------------------
<lambda>(["upload"], [".pioenvs/huzzah/firmware_00000.bin", ".pioenvs/huzzah/firmware_40000.bin"])
Auto-detected UPLOAD_PORT/DISK: /dev/cu.usbserial-AJ030SWR
"/Users/nouman/.platformio/packages/tool-esptool/esptool" -vv -cd ck -cb 115200 -cp /dev/cu.usbserial-AJ030SWR -ca 0x00000 -cf .pioenvs/huzzah/firmware_00000.bin -ca 0x40000 -cf .pioenvs/huzzah/firmware_40000.bin -cf .pioenvs/huzzah/firmware_00000.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to ck
setting baudrate from 115200 to 115200
setting port from /dev/tty.usbserial to /dev/cu.usbserial-AJ030SWR
setting address from 0x00000000 to 0x00000000
espcomm_upload_file
stat .pioenvs/huzzah/firmware_00000.bin success
opening port /dev/cu.usbserial-AJ030SWR at 115200
tcgetattr
tcsetattr
serial open
opening bootloader
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
[... 221 lines cut ...]
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
epc1=0x40243016, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000eb, depc=0x00000000
Fatal exception (28): 
```

NoumanSaleem, 2016-02-29:
For comparison, huzzah + arduino works

```ini
[env:huzzah]
platform = espressif
framework = arduino
board = huzzah

```

```c++
#include "Arduino.h"

void setup()
{
  Serial.begin(115200);
  Serial.print("SETUP FN()");
}

void loop()
{
}
```

ivankravets, 2016-02-29:
@NoumanSaleem thanks a lot for your [contributions](https://www.bountysource.com/teams/platformio). This is very complicated issue with ESP8266 SDK. However, we will try to resolve it.

Related issue #366

ivankravets, 2016-06-15:
Should be fixed in the latest release

closed 2016-06-15 by ivankravets