platformio/platformio-core #549: ESP8266: Espresso Lite v1.0 espcomm_send_command: cant receive slip payload data

ulno, 2016-02-29:
Hi all,

I am very excited about this project and was hoping to switch finally away from the Arduino IDE.

However, I can't upload to my ESP.
I get the following warning:
`espcomm_send_command: cant receive slip payload data
`
I am using PlatformIO 2.8.4 in Atom 1.5.3.
My serial converter is an ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC on /dev/ttyUSB0

It seems to be using esptool and not esptool.py.

Any way to switch to esptool.py or is this not necessary and I might have another problem?

The full upload output is:

```
"/home/ulno/.platformio/packages/tool-esptool/esptool" -vv -cd nodemcu -cb 115200 -cp /dev/ttyUSB0 -cf .pioenvs/espresso_lite_v1/firmware.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from /dev/ttyUSB0 to /dev/ttyUSB0
espcomm_upload_file
stat .pioenvs/espresso_lite_v1/firmware.bin success
opening port /dev/ttyUSB0 at 115200
tcgetattr
tcsetattr
serial open
opening bootloader
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
serialport_receive_C0: 00 instead of C0
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
serialport_receive_C0: 00 instead of C0
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 520 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 520 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
serialport_receive_C0: 00 instead of C0
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
serialport_receive_C0: 00 instead of C0
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 520 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
setting character timeout 0
scons: *** [upload] Error 255
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 520 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 520 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_sync failed
error: espcomm_open failed
```

valeros, 2016-02-29:
Hi @ulno Does upload via Arduino IDE work fine?

ulno, 2016-02-29:
Yes, it does. Any way, how I can try a different uploader or anything else I can do to debug?

valeros, 2016-02-29:
You can try [esptool.py](https://github.com/themadinventor/esptool):
Create new file `upload_extra_script.py` near `platformio.ini` with next content:
```
from SCons.Script import DefaultEnvironment
env = DefaultEnvironment()
env.Replace(
    NEWUPLOADER="path_to_esptool.py",
    NEWUPLOADERFLAGS=[
        "--port", "$UPLOAD_PORT",
        "--baud", "$UPLOAD_SPEED",
        "write_flash", "0x00000",
    ],
    UPLOADCMD='python $NEWUPLOADER $NEWUPLOADERFLAGS $SOURCE'
)

```

And specify extra script in your `platformio.ini`:

```
[env:espresso_lite_v1]
platform = espressif
framework = arduino
board = espresso_lite_v1
extra_script = upload_extra_script.py
```

ulno, 2016-03-01:
This works well - thanks a lot!

ulno, 2016-03-01:
I just upgraded PlatformIO to 2.8.5.dev2. I made sure my esp was reset (strangely I had to disconnect it first from power and then press the reset button). Then I closed the miniterm and then the upload proceeded.

Therefore, I tried back at the commandline (not in atom) with 2.8.4 and keeping a strict order (and removing power from the device before reset), I was able to program it. So assume that I was just not strict enough in making sure that the device was really reset.

Not sure what is going on here, but it seems to be a problem on my side.

Maybe an automatic close of the serial monitor (can probably hack this now thanks to your comment in my script) might help prevent this confusion.

closed 2016-03-01 by ulno

rousnay, 2017-01-16:
Try to connect The GPIO0 with a 10k resistor to ground and the CH_PD to VCC and it works :)