platformio/platformio-core #550: Arduino internal libraries not working

alangerak, 2016-02-29:
Recently, I upgraded from 2.6.1 to 2.8.4 and the internal libraries from Arduino are not working anymore. It doesn't recognise the default Arduino functions (digitalWrite etc.) but the arduino header(Arduino.h) is included.

I have run the tests using tox and the results are found [here](https://gist.githubusercontent.com/alangerak/2fc87d9f42ce2863406b/raw/5d32c12489ae2a891fae1418208658d0b91801e9/Results%2520of%2520platformio%2520test). (I had to use an exernal file, output is too large to show here)

I have tested this both Ubuntu 14.04 and Windows 8.1 but I had the same issue.

Is there a workaround for this problem other than replacing the internal libraries for external ones?

ivankravets, 2016-02-29:
What do you see in `~/.platformio/packages/framework-arduinoavr` ?

alangerak, 2016-02-29:
[Tree File structure](https://gist.githubusercontent.com/alangerak/2f2d2da83998fe6ab7c9/raw/2936b61dbf4c2d50fa01c4993c0f8b6f12754827/directories%2520of%2520framework%2520arduinoavr)

All the libraries seems to be there and I checked some of the libraries and there are identical to the ones that Arduino supplies with their IDE

ivankravets, 2016-02-29:
Try to remove `rm -rf ~/.platformio`.

alangerak, 2016-02-29:
So, basically reinstalling the toolchain right? In that case, it did not work. If you want to know, I have tried to link the created object file inside the library from the .pioenvs (inside of the project dir) to example code but gcc didn't complain about linking errors. Maybe there goes something wrong with linking to Arduino.h ?

ivankravets, 2016-02-29:
`.pioenvs` this is cache directory for PlatformIO Build System. Please don't depend on it. Try to remove this folder.

alangerak, 2016-03-01:
I have removed/reinstalled it, making sure that every file has deleted(including 'hidden directories'). After that, I tried the following: create [this project](https://gist.github.com/alangerak/78f862a835aae9d161dd) with megaatmega2560 as a board. When I run $platformio run , it will only give an error to the [map](https://www.arduino.cc/en/Reference/Map) function of Arduino. All other functions that does not use map, are compiling and working on the target board with servo. 

What is even more surprising (to me) is that if I add the following line at the end of the loop() functions:   map(0, 0, 1023, 0, 255);
The program does compile completely and works on the target board.

valeros, 2016-03-01:
Hi @alangerak 
I've just compiled your example without any error on Win8.1 x64.
Could you provide full building log?

alangerak, 2016-03-01:
You mean the terminal output of platformio run? [build log](https://gist.githubusercontent.com/alangerak/08e8f57dcbda107258c4/raw/1787801488320fb2665b4088900dd09b5bb0724a/log%2520file%2520platformio)

valeros, 2016-03-01:
@alangerak Please provide terminal output from `platformio run -t envdump`

alangerak, 2016-03-01:
[envdump](https://gist.githubusercontent.com/alangerak/b5e5519fbebcc5eea08e/raw/27fcc4a0d1ecc01acc699628299b3886bec9ec46/envdump%2520)

valeros, 2016-03-01:
Thanks, the problem is localized.
I created a special PlatformIO version for debugging, could you please test it and provide terminal output from `platformio run`?
`pip install https://github.com/platformio/platformio/archive/feature/issue-550-debug.zip`

alangerak, 2016-03-01:
[Terminal output](https://gist.githubusercontent.com/alangerak/70fad8b8758fd99e2559/raw/6407050ebbf339183ce967441dbd2f975841dca1/platformio%2520debug%2520run)

ivankravets, 2016-03-01:
Thanks a lot! Please try the latest http://docs.platformio.org/en/latest/installation.html#development-version

commit 8388359b (2016-03-01): Fix invalid detecting of compiler type // Resolve #550
closed 2016-03-01 by ivankravets