platformio/platformio-core #551: Programs don't run after todays update

jaroslavas, 2016-03-02:
Today, program asked me to update itself and some of the libraries. After the update platformio works weird - after I press upload, (I'm using ESP8266) it writes that it uploads but no dots are shown as usual, only after some time they appear in the console and console is closed immediately. After that, sketch doesn't run. If I compile program in Arduino IDE, everything works ok.

Atom version 1.5.3-3e71894
IDE 1.0.4 | CLI 2.8.4
Board: ESP8266-201

ivankravets, 2016-03-03:
Which was your previous version?

jaroslavas, 2016-03-03:
I'm not sure which version it was, but I downloaded it on February 23rd

ivankravets, 2016-03-03:
Please provide your `platformio.ini` file

valeros, 2016-03-03:
Hi @jaroslavas
Could you please also try to erase flash via `path/esptool -cp <port_name> -cd ck -ce` and than upload program again.

jaroslavas, 2016-03-03:
my platformio.ini:

```
[env:esp12e]
platform = espressif
framework = arduino
board = esp12e
```

ivankravets, 2016-03-03:
@jaroslavas please try https://github.com/platformio/platformio/issues/551#issuecomment-191690336

jaroslavas, 2016-03-03:
Are we speaking aboy [esptool.py](https://github.com/themadinventor/esptool) or [esptool-ck](https://github.com/igrr/esptool-ck)? Or maybe some other? :)

jaroslavas, 2016-03-03:
OK, it's the second one. And... it helped! Thank you!
Maybe you know what happened?

And one more question - now, immediately after the build and flashing ended, that part of the window flashes green and closes immediately, is it possible to make it stay?

valeros, 2016-03-03:
@jaroslavas Try `PlatformIO > Settings > Build Panel > Keep visible`

You mentioned above you are using `ESP8266-201` board which has 512Kb flash memory, but in your `platformio.ini` you chose `esp-12` board with 4Mb  flash. Is it OK for you?

ivankravets, 2016-03-04:
@jaroslavas please reopen this issue if you still have any problems

closed 2016-03-04 by ivankravets