platformio/platformio-core #552: Specific build_flags in [env:*] cause PlatformIO runtime error.

myrrlyn, 2016-03-02:
I've found a fault in how PlatformIO processes CCFLAGS-modifying options in the `build_flags` environment section of `platformio.ini`.

I've attached a brief overview of my system, my `platformio.ini` file, and the log of `platformio run`'s output.

https://gist.github.com/myrrlyn/c1f3a2fd5e6a8d0ef613

ivankravets, 2016-03-03:
* If you use PlatformIO IDE, please click on Gear(Settings) icons, Community Packages > platformio-ide > Settings > Use development version of PlatformIO.

Please re-test it.

ivankravets, 2016-03-03:
Also, `build_flags = -include Arduino.h` will raise error. That is OK, because you are going "to insert" into EACH build file "Arduino.h" which raises conflicts.

commit 4c4682f4 (2016-03-03): Fix incorrect parsing of GCC "-include" flag // Resolve #552
closed 2016-03-03 by ivankravets