platformio/platformio-core #555: Attemping `pio update` fails at espressif

D3f0, 2016-03-03:
OS: OS X 10.11.3
Python: 2.7.11 (from Homebrew)

```bash
$ pio update

Platform atmelavr
--------
Updating toolchain-atmelavr package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating tool-avrdude package:
Versions: Current=2, Latest=2    [Up-to-date]
Updating framework-arduinoavr package:
Versions: Current=26, Latest=26      [Up-to-date]
Updating tool-micronucleus package:
Versions: Current=1, Latest=1    [Up-to-date]

Platform espressif
--------
Updating toolchain-xtensa package:
Versions: Current=2, Latest=2    [Up-to-date]
Updating tool-esptool package:
Versions: Current=5, Latest=5    [Up-to-date]
Updating tool-mkspiffs package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating framework-arduinoespressif package:
Versions: Current=8, Latest=10   [Out-of-date]
Uninstalling framework-arduinoespressif package:    Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/update.py", line 26, in cli
    ctx.invoke(cmd_platforms_update)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/platforms.py", line 169, in platforms_update
    p.update()
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 369, in update
    pm.update(name)
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 153, in update
    self.uninstall(name)
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 129, in uninstall
    rmtree(join(self._package_dir, name))
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 239, in rmtree
    onerror(os.listdir, path, sys.exc_info())
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 237, in rmtree
    names = os.listdir(path)
OSError: [Errno 2] No such file or directory: '/Users/nahuel/.platformio/packages/framework-arduinoespressif'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

ivankravets, 2016-03-03:
Thanks! Please test it in http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-03-03 by ivankravets

D3f0, 2016-03-03:
@ivankravets Worked like a charm!
Thank you for such a quick response!

```bash
~ » pip uninstall platformio
Uninstalling platformio-2.8.4:
  /usr/local/bin/pio
  /usr/local/bin/platformio
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/DESCRIPTION.rst
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/INSTALLER
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/METADATA
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/RECORD
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/WHEEL
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/entry_points.txt
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/metadata.json
  /usr/local/lib/python2.7/site-packages/platformio-2.8.4.dist-info/top_level.txt
  /usr/local/lib/python2.7/site-packages/platformio/__init__.py
  /usr/local/lib/python2.7/site-packages/platformio/__init__.pyc
  /usr/local/lib/python2.7/site-packages/platformio/__main__.py
  /usr/local/lib/python2.7/site-packages/platformio/__main__.pyc
  /usr/local/lib/python2.7/site-packages/platformio/app.py
  /usr/local/lib/python2.7/site-packages/platformio/app.pyc
  /usr/local/lib/python2.7/site-packages/platformio/boards/adafruit.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/arduino.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/atmelsam.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/digistump.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/engduino.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/espressif.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/freescalekinetis.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/microduino.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/misc.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/nordicnrf51.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/nxplpc.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/punchthrough.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/raspberrypi.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/siliconlabsefm32.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/sparkfun.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/ststm32.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/teensy.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/timsp430.json
  /usr/local/lib/python2.7/site-packages/platformio/boards/titiva.json
  /usr/local/lib/python2.7/site-packages/platformio/builder/__init__.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/__init__.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/main.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/main.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/__init__.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/__init__.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/atmelavr.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/atmelavr.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/atmelsam.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/atmelsam.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/basearm.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/basearm.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/baseavr.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/baseavr.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/espressif.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/espressif.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/__init__.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/__init__.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/arduino.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/arduino.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/cmsis.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/cmsis.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/energia.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/energia.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/libopencm3.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/libopencm3.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/mbed.py
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/mbed.pyc
  /usr/local/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/spl.py
[... 207 lines cut ...]

Platform espressif
--------
Updating framework-arduinoespressif package:
Versions: Current=10, Latest=10 	 [Up-to-date]
Updating ldscripts package:
Versions: Current=19, Latest=19 	 [Up-to-date]

Platform nxplpc
--------
Updating framework-mbed package:
Versions: Current=11, Latest=11 	 [Up-to-date]
Updating toolchain-gccarmnoneeabi package:
Versions: Current=2, Latest=2 	 [Up-to-date]

Updating [ 121 ] SerialCLI library:
Versions: Current=9186338579, Latest=754f608e08 	 [Out-of-date]
The library #121 'SerialCLI' has been successfully uninstalled!
Installing library [ 121 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #121 'SerialCLI' has been successfully installed!
Updating [ 89 ] PubSubClient library:
Versions: Current=2.6, Latest=2.6 	 [Up-to-date]
```

ivankravets, 2016-03-03:
Thanks! Don't forget to upgrade to stable version when it will be released.

commit 1e3e3dd9 (2016-03-03): Fix issue with updating package which was deleted manually by user // Resolve #555