platformio/platformio-core #557: Crawler not updating my lib?

marvinroger, 2016-03-04:


ivankravets, 2016-03-04:
Please provide link to your library in our Registry.

marvinroger, 2016-03-04:
@ivankravets sorry, of course I provided the link, but it seems like the content of the issue disappeared. I updated the first post. ;)

ivankravets, 2016-03-04:
Crawlers syncronizes lib 1 time per 24 hours. I see a few problems with your lib:

1. You registered lib with version 1.0.0
2. Then downgraded to 0.1.0
3. Today, you again release as 1.0.0

I fixed this issue in DB. Please be careful with versioning.

ivankravets, 2016-03-04:
See updated Registry http://platformio.org/#!/lib/show/555/Homie

marvinroger, 2016-03-04:
Yes, I messed up. Thanks for the correction.
What about an hook (GitHub hook in our case) instead of pooling repos every 24h? This would allow for faster updates on registry and would probably reduce bandwidth server side. This could land in pio 3?

ivankravets, 2017-01-05:
Moved to https://github.com/platformio/platformio-api/issues/23

closed 2017-01-05 by ivankravets