platformio/platformio-core #558: MySensors Gateway example not compiling

gerritv, 2016-03-04:
Attached is a project directory which fails to build using `platformio` but builds ok using VisualMicro and Arduino IDE 1.6.7.
Also attached are two build outputs ( from Visual Micro and other from platformio) and the envdump.
Platformio version is 2.8.5.dev2 running on Windows 10 with python 2.7. The problem arises when building for Arduino nano as well, doesn't seem ESP8266 specific. My guess is that there is a flaw in the convoluted Arduino file finding algorithm. Platformio seems to want to build files that are not relevant to the .ino in question.

[SensorNet.zip](https://github.com/platformio/platformio/files/158823/SensorNet.zip)
[envdump.txt](https://github.com/platformio/platformio/files/158824/envdump.txt)

[PlatformioOutput.txt](https://github.com/platformio/platformio/files/158825/PlatformioOutput.txt)
[VisualMicroOutput.txt](https://github.com/platformio/platformio/files/158827/VisualMicroOutput.txt)

valeros, 2016-03-04:
Hi @gerritv 
Indeed, MySensors library has an unusual build logic. Please have a look at #331 
I've created a working [project](https://dl.dropboxusercontent.com/u/10025728/mysensors.zip) for you, could you test it?

gerritv, 2016-03-04:
Awesome, it works. I guess next time I see these symptoms I will have to try this method. Sadly Arduino projects seem to provoke some really obtuse habits in structure :-(

closed 2016-03-04 by gerritv