platformio/platformio-core #560: Solution if terminal-plus is already installed

brbrr, 2016-03-05:
I have already installed terminal-plus package. Is there any way to select which terminal to use?

ivankravets, 2016-03-05:
The original `terminal-plus` is not compatible with PlatformIO IDE. I propose you to use `platformio-ide-terminal` which we maintain now.

ivankravets, 2016-03-05:
I want to note, that `platformio-ide-terminal` has the same functionality that original package + additional enhancements. You can use single terminal for both cases.

closed 2016-03-05 by ivankravets

brbrr, 2016-03-05:
May I know what the difference with `terminal-plus' ?

ivankravets, 2016-03-05:
@brbrr please take a look at https://github.com/platformio/platformio-atom-ide-terminal/blob/master/CHANGELOG.md

brbrr, 2016-03-05:
Ah, of course :sweat_smile:
Thanks!

ivankravets, 2016-03-05:
In a nutshell, API to pass own commands to it + refactored "status panel" which was moved to native Atom's status bar (saved 30px and empty line ) :)

ivankravets, 2016-03-05:
What is more, we are glad to move our improvements to original terminal. See https://github.com/jeremyramin/terminal-plus/pull/160

However, @jeremyramin doesn't support this project and no one has access to it :(

brbrr, 2016-03-05:
Ah, I see. there definitely lack of activity in original project.