platformio/platformio-core #561: Serial monitor crashes on receiving junk from ESP8266

brbrr, 2016-03-05:
Using arduino nano with ESP8266 connected, on hard reset step - terminal session ends up with error:

```
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/brbrr/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 411, in reader
    self.console.write(text)
  File "/Users/brbrr/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 60, in write
    self.output.write(s)
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufffd' in position 0: ordinal not in range(128)
```

ivankravets, 2016-03-05:
Could you call Serial Monitor directly from PlatformIO IDE Terminal without GUI? I mean
```
pio serialports monitor
```

brbrr, 2016-03-05:
I got same errors. btw, termination signals is not working for me(Ctrl+C/X)
Forgot to mention, Im using OS X

ivankravets, 2016-03-05:
Please try to play with http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#cmdoption-platformio-serialports-monitor-f
```
pio serialports monitor -f direct
```
> termination signals is not working for me(Ctrl+C/X)

That is OK. Use `Ctrl+]`. See http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#cmdoption-platformio-serialports-monitor--exit-char

brbrr, 2016-03-05:
hm, in Terminal.app `$ pio serialports monitor` works just fine, but in pio terminal it fails. manage it working with '$ pio serialports monitor --raw'

So it's kinda pio teminal issue?

ivankravets, 2016-03-05:
Could provide sketch to reproduce this issue?

brbrr, 2016-03-05:
```
#include <SoftwareSerial.h>

#define ESP_RST   5
SoftwareSerial esp(10, 11);
#define debug Serial

void setup() {
  esp.begin(9600);
  debug.begin(9600);
  digitalWrite(ESP_RST, LOW);
  pinMode(ESP_RST, OUTPUT);
  delay(100);
  pinMode(ESP_RST, INPUT);
    digitalWrite(ESP_RST, HIGH);

  unsigned long time = millis();
  time += 6000;
  do {
    if (esp.available())
      Serial.write(esp.read());
  }
  while (millis() < time);
}

void loop() {
}
```

ivankravets, 2016-03-05:
Thanks, I'll take a look on it. 

Temporary, please use system Terminal.

Please, sorry for the troubles.

tzapu, 2016-03-25:
thanks for your wonderful platform, i am having the same issue, in my case it crashes at boot (probably due to the 78440 baud rate the esp8266 uses when booting
```
pio serialports monitor -b 115200
--- Miniterm on /dev/cu.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
rException in thread rx:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/alex/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 411, in reader
    self.console.write(text)
  File "/Users/alex/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 60, in write
    self.output.write(s)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2400' in position 1: ordinal not in range(128)
```

works with --raw

thanks again

ivankravets, 2016-03-26:
Moved to PlatformIO IDE project https://github.com/platformio/platformio-atom-ide/issues/84

closed 2016-03-26 by ivankravets

brbrr, 2016-03-26:
btw, it works with latest(dev) version

ivankravets, 2016-03-26:
@brbrr Sorry, what do you mean under `latest(dev)`? For PlatformIO? PySerial?

brbrr, 2016-03-26:
dev platformIO-IDE

ivankravets, 2016-03-26:
@brbrr Please uninstall PlatformIO IDE packages and install the latest 1.1.0. Do you have problems with it?

carl3, 2016-11-25:
I had the same problem today. New version of PlatformIO and the serial was too old for the IDE and needed to be rebuilt (package manager said it was installed, but didn't work. Clicking red bug icon in lower left gave the instructions to rebuild then OK.)

The problem I had was a wrong baud rate setting (from the example code). If miniterm.py gets any bad character, it dies a horrible death. It would be better to substitute some weird character if there is an encoding problem rather than crash. It's not unusual to get garbled characters