platformio/platformio-core #563: Loop in atom updater

xxv, 2016-03-05:
Each time I start atom (for platformio), I get a window that pops up saying:

> PlatformIO IDE has been successfully installed!
> 
> However, some of its components will only become available after Atom window reload. You can click "Reload now" button below to perform reload immediately, or click "Reload later" and perform reload yourself with "View > Reload" command whenever you're ready.

with a "Reload now" and "Reload later" button. If I reload now, it restarts the IDE and then pops up again. If I press "Reload later" it goes away until I restart atom.

Versions: IDE 1.0.4 | CLI 2.8.4

Running on Debian stretch/sid

ivankravets, 2016-03-05:
Please move issue to appropriate repository https://github.com/platformio/platformio-atom-ide

closed 2016-03-05 by ivankravets

xxv, 2016-03-05:
Whoops, sorry about that. Moved here: https://github.com/platformio/platformio-atom-ide/issues/63