platformio/platformio-core #564: Deploying Arduino Projects to Bare AVRs (no bootloader) Fails Verification. Requesting Board Definition for Bare AVR

neilmovva, 2016-03-06:
While PlatformIO handles the Arduino build system perfectly, it seems we cannot use an ISP (in-system programmer, eg. USBasp, AVR Dragon, etc.) to upload the firmware onto an AVR without the Arduino bootloader. The flash is written, but verification fails and the program does not execute as expected (sometimes producing entirely undefined behavior!) 

The issue seems to be that PlatformIO assumes the presence of the Arduino bootloader, and writes the firmware to flash with a bootloader-sized offset (from 0.5 to 4 kB, depending on the board selected). There's also a hint in support of this theory in the board definitions; none of the AVR chips seem to have the "correct" amount of flash program space - they're all missing bytes the size of the associated bootloader (e.g. the ATMega328P's range from 30kB to 31.5kB, which correspond to various bootloaders commonly shipped on Arduinos). 

Unfortunately, using the ISP programmer seems to incorrectly account for the bootloader offset, and the program is written somewhere it shouldn't be. Post-write verification fails, too.  I think there's an easy fix in defining a new class of boards which have no bootloader magic at all, but still support the Arduino ecosystem all the same (the bootloader should have nothing to do with code library support). I think there's a strong use case in PlatformIO's audience for using the Arduino library system with more flexible hardware (general AVR chips). I did attempt to find and bypass the bootloader business, but quickly got lost in the build/config scripts. Would appreciate any guidance, and would be happy to help test and develop a solution.

Thanks for all your great work. PlatformIO is a very refreshing experience in the world of embedded!

ivankravets, 2016-03-06:
Have you read this http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer?

Using `upload_flags` you can specify own flags to `avrdude`. Do you know what should be changed from our side?

neilmovva, 2016-03-07:
Ivan, you're great - it was a simple matter of changing the flags (specifically, removing the -D option so that the chip is erased for each program cycle). There is no issue with PlatformIO. Thanks for identifying the issue so quickly, I should have read the docs a little closer. 

The extra_script functionality you've built is especially useful - anyone with similar issues down the line might refer to [the extra_script docs](http://docs.platformio.org/en/latest/projectconf.html#extra-script), as well as the template from Ivan in #247.  

Thanks again for your work!

closed 2016-03-07 by neilmovva

skorokithakis, 2016-04-27:
@neilmovva Can you post your options, for posterity? I'm having the same problem and it would be much more helpful if I could see what you did.

ivankravets, 2016-04-27:
@skorokithakis please try "Upload using Programmer" instead "Upload" (`pio run -t program`)

ivankravets, 2016-04-27:
@skorokithakis docs http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

skorokithakis, 2016-04-27:
Ah, that simple? Will do, thanks. I just need to figure out how to connect these unlabelled pins to the FTDI first.

neilmovva, 2016-04-27:
@skorokithakis Here's my extra_script.py:

```
from os.path import join
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(
    MYUPLOADERFLAGS=[
        "-v",
        "-p", "$BOARD_MCU",
        "-C",
        '"%s"' % join("$PIOPACKAGES_DIR", "tool-avrdude", "avrdude.conf"),
        "-c", "$UPLOAD_PROTOCOL",
     ],
    UPLOADHEXCMD='"$UPLOADER" $MYUPLOADERFLAGS -U flash:w:$SOURCES:i'
)
```
hope that helps!

skorokithakis, 2016-04-27:
@neilmovva It does, thank you!

ivankravets, 2016-04-27:
@neilmovva why do you need this extra script? The `pio run -t program` should work as well.