platformio/platformio-core #565: Unexpected results with Print class

sterling, 2016-03-06:
### Configuration

**Operating system**: Mac OSX 10.11.3

**PlatformIO Version** (`platformio --version`): 2.8.4

### Description of problem
When inheriting the Print class, behavior differs between the Arduino IDE and PlatformIO IDE. Passing in a `const char*` to `print` only calls `write` with the first character when using PlatformIO. When using Arduino IDE, write is called for each character of the string. 

#### Steps to Reproduce

1. Load and run the provided file below onto an Arduino UNO using PlatformIO IDE
2. Then load and run the same file using the Arduino IDE

### Actual Results
PlatformIO IDE:
```
H
```
Arduino IDE: 
```
H
e
l
l
o
```

### Expected Results
PlatformIO should match Arduino IDE

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
```

**Source file to reproduce issue:**
```cpp
#include "Arduino.h"

class PrintTest : public Print {
public:
  virtual size_t write(uint8_t data) {
    Serial.println((char)data);
    return 0;
  }
} PrintTest;

void setup() {
  Serial.begin(9600);
  PrintTest.print("Hello");
}
void loop() {}
```

valeros, 2016-03-07:
Hi @sterling 
I tried your example in the Arduino 1.6.7, and it also prints only one `H` symbol.
Could you please try next code:
```
#include "Arduino.h"

class PrintTest : public Print {
public:
  virtual size_t write(uint8_t data) {
    return Serial.println((char)data);
  }
} PrintTest;

void setup() {
  Serial.begin(9600);
  PrintTest.print("Hello");
}
void loop() {}
```

sterling, 2016-03-07:
@valeros Thanks for looking into this. 

I've tried your updated code and I am getting the same results (outputs the full string). I've also updated to Arduino 1.6.7 because I was on 1.6.5, but that has not changed my results. I will try a fresh install on another machine to see what results I get.

sterling, 2016-03-07:
I just tried the code on a fresh install of Arduino 1.6.7 on Mac OSX and I got the same results. The arduino IDE printed out the full string.

sterling, 2016-03-07:
Actually, after trying your updated code on PlatformIO, it is now outputting the entire string. It's odd that my Arduino IDE does not respect the return value.

ivankravets, 2016-03-08:
@sterling does it mean that PlatformIO works well?

sterling, 2016-03-09:
@ivankravets I believe PlatformIO is probably behaving correctly. The only weird thing is that I am not getting the same results between IDEs. Can anyone reproduce what I am seeing?

orgkhnargh, 2016-03-09:
@sterling I tried your code on Arduino IDE 1.6.7 and PlatformIO IDE. Both print only first character, which appears to be the correct behavior, because `write(const uint8_t *buffer, size_t size)` works until ` write(uint8_t data)` returns `false`. Check the [source code](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Print.cpp#L38). You made `write(uint8_t data)` to return `0` every time, so `write(const uint8_t *buffer, size_t size)` returns after the first iteration. Sounds about right to me.

Please check again. Your code should not work as you expect in Arduino IDE either.

sterling, 2016-03-09:
You're right, Arduino 1.6.7 only outputs a single character. I must have tested it incorrectly when I updated. I just discovered that this behavior changes sometime between these two versions (1.6.7 and 1.6.5). The older version outputs the entire string, while the newer does not. So PlatformIO is working exactly as expected. Thanks for the help everyone.

closed 2016-03-09 by sterling

orgkhnargh, 2016-03-09:
We're glad to help! We hope you enjoy the PlatformIO IDE!