platformio/platformio-core #566: Large Teensy 3.1 project builds but does not upload or run

blackketter, 2016-03-08:
### Configuration
MacOS El Capitan
PlatformIO, version 2.8.5, installed via homebrew

### Description of problem
Building a large (many libraries and code files) Teensy 3.1 project outputs a meaningfully different elf file and hex file as compared to a working one built using Arduino 1.6.7/Teensyduino.
The hex file fails to upload via the CLI uploader, but does seem to work in the GUI uploader.
Once uploaded the app fails to start, no debug output.

A trivial blink app (using the same platformio.ini file below) works fine on the same system.

#### Steps to Reproduce

1.  Build app via Arduino/Teensyduino. Works.
2.  Build app via PlatformIO, generates different elf and bad hex file

### If problems with PlatformIO Build System:
Some warnings but no errors in build results below.

**The content of `platformio.ini`:**
```ini
[env:teensy31]
lib-ignore=extras
platform = teensy
framework = arduino
board = teensy31
build_flags = -DTEENSY31 -UUSB_SERIAL -DTEENSYDUINO=127 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH 
board_f_cpu = 96000000
```

**Source file to reproduce issue:**
build results:
[buildresults.txt](https://github.com/platformio/platformio/files/164181/buildresults.txt)
.elf and .hex files from platformio:
[firmware.zip](https://github.com/platformio/platformio/files/164186/firmware.zip)
Working elf and hex files as built by Arduino
[working.zip](https://github.com/platformio/platformio/files/164189/working.zip)
Source files available if you need them, it's a pretty big project.

ivankravets, 2016-03-09:
@blackketter I have the same error 
```
"/Users/ikravets/.platformio/packages/tool-teensy/teensy_loader_cli" -mmcu=mk20dx256 -w -s -v .pioenvs/teensy31/firmware.hex
Teensy Loader, Command Line, Version 2.0
error reading intel hex file ".pioenvs/teensy31/firmware.hex"
Warning, HEX parse error line 8193
scons: *** [uploadlazy] Error 1
```

Could you provide Build Output from Teensyduino? I remember that in Arduino IDE you can enable verbose mode.

------

@PaulStoffregen, do you have any ideas?

ivankravets, 2016-03-09:
@blackketter I've just tried your WORKING hex from Teensyduino and it also doesn't work:
```
"/Users/ikravets/.platformio/packages/tool-teensy/teensy_loader_cli" -mmcu=mk20dx256 -w -s -v .pioenvs/teensy31/BritepadSketch.ino.hex
Teensy Loader, Command Line, Version 2.0
Warning, HEX parse error line 8193
error reading intel hex file ".pioenvs/teensy31/BritepadSketch.ino.hex"
```

Looks like we have problem with Teensy CLI.

P.S: Could you try to upload PlatformIO's firmware using Teensy Loader GUI?

valeros, 2016-03-09:
@blackketter It would be great if you could send me (valeros08@gmail.com) your project to reproduce the issue.

blackketter, 2016-03-09:
Here are the teensyduino build results for a working build:
[teensyduino-buildresults.txt](https://github.com/platformio/platformio/files/165546/teensyduino-buildresults.txt)

And here is a zip of the full project, including libs:
[bp-platformio.zip](https://github.com/platformio/platformio/files/165569/bp-platformio.zip)

blackketter, 2016-03-09:
@PaulStoffregen @ivankravets Shall I open a new bug regarding the teensy_loader_cli?  If so, where?

PaulStoffregen, 2016-03-09:
here:
https://github.com/PaulStoffregen/teensy_loader_cli

blackketter, 2016-03-09:
@ivankravets Upon further testing, the latest version of teensy_loader_cli from @PaulStoffregen's repo works fine:

```
./teensy_loader_cli -mmcu=mk20dx256 -w -s -v BritepadSketch.ino.hex                                                                               
Teensy Loader, Command Line, Version 2.0
Read "BritepadSketch.ino.hex": 212236 bytes, 81.0% usage
Soft reboot is not implemented for OSX
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read "BritepadSketch.ino.hex": 212236 bytes, 81.0% usage
Programming................................................................................................................................................................................................................
Booting
```

The one in platformio does not:

```
~/.platformio/packages/tool-teensy/teensy_loader_cli -mmcu=mk20dx256 -w -s -v BritepadSketch.ino.hex 
Teensy Loader, Command Line, Version 2.0
Warning, HEX parse error line 8193
error reading intel hex file "BritepadSketch.ino.hex"
```
I've opened bug #568

The remaining problem of the hex file not being built right remains.

ivankravets, 2016-03-09:
> The remaining problem of the hex file not being built right remains.

Do you mean that PlatformIO's firmware doesn't work?

blackketter, 2016-03-09:
That's right.  The bug as reported was actually two bugs:

1. PlatformIO's teensy_loader_cli doesn't work for this project. This issue has been moved to bug #568 
2. The [firmware](https://github.com/platformio/platformio/files/164186/firmware.zip) built by platformio appears to be substantially different from the [one built by Teensyduino](https://github.com/platformio/platformio/files/164189/working.zip).  Once uploaded with a working uploader it does not start.

Those archives include .hex and .elf.  

The complete PlatformIO project is [here](https://github.com/platformio/platformio/files/165569/bp-platformio.zip).

PaulStoffregen, 2016-03-09:
A bug was recently fixed in teensy_loader_cli.

Perhaps PlatformIO has an older version, before the recent fix?

blackketter, 2016-03-13:
The teensy_loader_cli has been updated and works great.

The issue remains that the firmware built by platformio does not work. I am able to build, upload and run a trivial blink app, but the project included above builds and uploads, but does not run.

I'm not sure where to go from here.

ivankravets, 2016-03-13:
@valeros is working on it. I hope we will have some results tomorrow.

blackketter, 2016-03-13:
Awesome, thanks Ivan.

PaulStoffregen, 2016-03-14:
Not sure if I can really do anything to help... but if there's any sort of problem on the Teensy, please let me know?

ivankravets, 2016-03-14:
@PaulStoffregen I hope we will resolve it soon. Thanks a lot that you are ready to help us. 

I forgot to mention, that PlatformIO Build System doesn't depend on Arduino/Teensyduino. We have own build workflow. PlatformIO builds each library to static archive `*.a` and then links it with the rest lib archives and objects from the source project. However, Arduino IDE doesn't work with archives, it links whole bunch of objects directly.

We have a lot of Teensy users in our community. This is the first serious problem when firmware from PlatformIO doesn't work. Of course, this is a problem of PlatformIO Build System, but not of Teensy board or Arduino Core. 

Thanks @blackketter that posted this issue with the source code. It helps us to make our Build System more compatible with Arduino IDE/Teensyduino.

valeros, 2016-03-15:
@blackketter 
I reproduced your problem. Indeed, a firmware doesn't run after flashing for some reason.
Can I ask you to do a small experiment with your project in Arduino IDE?
Please rename your file `BritepadSketch.ino` to `ZBritepadSketch.ino` and upload firmware to your board.
Does it work OK?

blackketter, 2016-03-15:
@valeros  Very interesting:

1.  I created a new directory called ZBritepadSketch and copied the .cpp, .h and .ino files to it, renamed BritepadSketch.ino to ZBritepadSketch.ino.  It successfully built and uploaded but did not run.

2. I renamed the file and directory back to BritepadSketch and it  built successfully and ran.

What does this mean?

valeros, 2016-03-15:
Looks like the problem appears on the linking stage due to object files order.
For now, I can't say what is wrong, but maybe @PaulStoffregen could help us?
As a workaround, you can rename your file `BritepadSketch.ino` to `BritepadSketch.cpp` and it should work properly with PlatformIO.

blackketter, 2016-03-15:
Awesome.  That worked.  Let me know if there's anything else I can do to help resolve the root issue.

ivankravets, 2016-03-15:
@blackketter please wait for @PaulStoffregen. 

Nevertheless, that is great signal for PlatformIO Build System that your huge and complicated project works. As I've mentioned above, we use own build workflow and don't depend on Arduino builder. What is more, our Build System is 3-4x faster than Arduino.

ivankravets, 2016-03-15:
@blackketter Have tried our IDE? http://platformio.org/platformio-ide

blackketter, 2016-03-15:
Thanks!  I'm really excited about platformio.  I will try the platformio IDE, but I generally prefer working from the command line. I'll be happy to retire the Arduino IDE for sure!

PaulStoffregen, 2016-03-15:
I'm hoping to look at the IDE this summer.

blackketter, 2016-03-15:
That workaround, naming "BritepadSketch.ino" to "BritepadSketch.cpp" works.  Of course, this breaks building under Arduino.

@PaulStoffregen Any idea why the order of the linked files matters?

PaulStoffregen, 2016-03-16:
Sorry, I can't look into this.  Way too much other stuff to do.  So I really can't offer any specific advice, other than the general guess that perhaps your code and something in a .a library conflict, or symbols in multiple .a libs might conflict.  When normal .o has conflicting symbols, the linker quits with an error.  When in .a libs, the first on the list overrides the others.

ivankravets, 2016-03-16:
@PaulStoffregen the problem is not linked with PlatformIO. Teensyduino keeps the original name when converting *.ino to *.cpp. For example, in this project BritepadSketch.ino will be converted to BritepadSketch.ino.cpp. Of course, it works... If user renames BritepadSketch.ino to Teensy.ino it will not work even with Teensyduino.

Why it doesn't work with PlatformIO without any changes to BritepadSketch.ino? It doesn't work because PlatformIO convert `BritepadSketch.ino` to `tmp_ino_to.cpp`. As result, the order for linker is different.

-----

@blackketter please copy this issue here https://github.com/PaulStoffregen/Arduino-1.6.8-Teensyduino and make back reference to this issue.

ivankravets, 2016-04-09:
I close this issue.

@blackketter if this issue will be fixed in the new Teensyduino, please report us and we will update it.

closed 2016-04-09 by ivankravets