platformio/platformio-core #567: Add option to ignore some library subdirectories

blackketter, 2016-03-09:
### Configuration

**Operating system**: 
MacOS El Capitan
Teensy3.1
PlatformIO, version 2.8.5

### Description of problem
Some Arduino libraries include an "extras" directory with useful tools that should not be compiled into a project.  An example is https://github.com/PaulStoffregen/ILI9341_t3

When this library (or a locally modified version) is included in libs, the contents of "extras" tries to be built, but fails.  Deleting or moving this subdirectory works fine.

It would be great to have a platformio.ini directive for ignoring some subdirs.  

Possibly have the lib-ignore directive match on subdirectories too, like:
lib-ignore=*/extras
or the like...

ivankravets, 2016-03-09:
How did you install this library? See https://github.com/PaulStoffregen/ILI9341_t3/blob/master/library.json

Please try this library http://platformio.org/#!/lib/show/540/ILI9341_t3

blackketter, 2016-03-09:
Thanks, Ivan.  I have a newer version with local modifications checked out into my lib folder.  I understand that the platformio version excludes the "extras" folder, I'd like a way to do that for libs included with my project.

ivankravets, 2016-03-09:
Dear Dean,

We have opened issue https://github.com/platformio/platformio/issues/289 where developers of libs be able to control Build System and specify which source code should be built.

Regards Ivan.

closed 2016-03-09 by ivankravets