platformio/platformio-core #568: teensy_loader_cli in platformio breaks on some large .hex files

blackketter, 2016-03-09:
### Configuration
MacOS X El Capitan
PlatformIO, version 2.8.5
Teensy Loader, Command Line, Version 2.0

### Description of problem
The attached Teensy3.1 hex file:

[BritepadSketch.ino.hex.zip](https://github.com/platformio/platformio/files/166034/BritepadSketch.ino.hex.zip)

does not upload via the teensy_loader_cli as built by platformio.

A fresh build from https://github.com/PaulStoffregen/teensy_loader_cli succeeds.

### Actual Results
```
~/.platformio/packages/tool-teensy/teensy_loader_cli -mmcu=mk20dx256 -w -s -v BritepadSketch.ino.hex 
Teensy Loader, Command Line, Version 2.0
Warning, HEX parse error line 8193
error reading intel hex file "BritepadSketch.ino.hex"
```
### Expected Results
```
./teensy_loader_cli -mmcu=mk20dx256 -w -s -v BritepadSketch.ino.hex                                                                               
Teensy Loader, Command Line, Version 2.0
Read "BritepadSketch.ino.hex": 212236 bytes, 81.0% usage
Soft reboot is not implemented for OSX
Waiting for Teensy device...
 (hint: press the reset button)
Found HalfKay Bootloader
Read "BritepadSketch.ino.hex": 212236 bytes, 81.0% usage
Programming................................................................................................................................................................................................................
Booting
```

ivankravets, 2016-03-09:
@blackketter please provide `teensy_loader_cli` binary

blackketter, 2016-03-09:
Here's my local build:

[teensy_loader_cli.zip](https://github.com/platformio/platformio/files/166043/teensy_loader_cli.zip)

ivankravets, 2016-03-09:
@valeros please rebuild Teensy CLI for all paltforms