platformio/platformio-core #570: BuildDependentLibraries ignores src_filter

maru-sama, 2016-03-10:
I recently started using the src_filter directive to create two different firmwares in one project. Both of them are runing on the same platform but have a different functionality and use different libraries. While compiling one of the environments I noticed that Libraries were compiled which were referenced in the other main file but not in the current file. Looking at https://github.com/platformio/platformio/blob/4c4682f45709ad54da5293eb654bff9bbcc05c44/platformio/builder/tools/platformio.py#L224 I saw that BuildDependentLibraries apparently does not use src_filter. While this does not affect the final firmware binary it increases the build time.

```
[env:display]
platform = espressif
framework = arduino
board = huzzah
upload_speed = 921600
upload_port = "ESP8266-Lights.local"
src_filter = "+<display>"

[env:trigger]
platform = espressif
framework = arduino
board = huzzah
upload_speed = 921600
upload_port = "ESP8266-Trigger.local"
src_filter = "+<trigger>"
```

If I compile trigger here the main.cpp file in display is actually parsed as well and all the libs referenced there are also build, although they are not referenced in the main.cpp file inside trigger.

ivankravets, 2016-03-10:
Do you mean this functionality https://github.com/platformio/platformio/issues/289?

maru-sama, 2016-03-10:
I do not think it is related. Let me try to explain this in more detail. In the above example I have to folders below src with one main.cpp file each. I used src_filter to be able to select the main.cpp file I want to compile and this works quite well. The main.cpp files are using different libraries. the one in display is using the FastLed library for example. If I now build the trigger environment which is NOT using the fastled library it is still being compiled during the build process. I think the reason for this is that BuildDependentLibraries is looking at ALL files in the source folder, finds both main.cpp files, extracts the includes and then builds the libraries.

ivankravets, 2016-03-10:
@maru-sama Thanks, I got you now. This is a bug.

ivankravets, 2016-03-10:
I want to note that if library is not used in source code but IS BUILT and LINKED, GCC Linker will ignore it because there doesn't exist any reference to it.

Of course, PlatformIO should ignore them. I mean that your code should work well with the current version of PlatformIO.

maru-sama, 2016-03-10:
I had a look at the linking command and apparently it is not linked in either. So apparently this logic is fine as well. Right now this is really only affecting the build-time.

ivankravets, 2016-03-10:
I've jus changed it to "enhancement" and will fix together with the rest lib-related issues in PlatformIO 3.0. PlatformIO 2.0 is very stable and I don't want to break it. Library Dependencies Finder (LDF) has complicated logic.

maru-sama, 2016-03-10:
Perfect

ivankravets, 2016-06-22:
Resolved in https://github.com/platformio/platformio/commit/f8db1d11a703d5ebcc41a621109c4dc46046ac5b

closed 2016-06-22 by ivankravets