platformio/platformio-core #571: Getting multiple definition errors when I try and compile code for mbed platform

gilandose, 2016-03-10:
**Operating system**:
Linux
PlatformIO Version 2.8.5

### Description of problem
When I compile against a crypto library for the mbed platform it fails with the multiple definitions errors as seen below. The code compiles correctly in the online mbed compiler. Am happy to try out some things to fix it with some guidance. Crypto library can be found [here](https://developer.mbed.org/users/feb11/code/Crypto/)

### Actual Results
> /home/rgilmore/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/softfp/libstdc++_s.a(vterminate.o): In function `__gnu_cxx::__verbose_terminate_handler()':
vterminate.cc:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): multiple definition of `__gnu_cxx::__verbose_terminate_handler()'
.pioenvs/nucleo_f401re/libFrameworkMbed.a(retarget.o):retarget.cpp:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): first defined here
/home/rgilmore/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/softfp/libstdc++_s.a(atexit_arm.o): In function `__aeabi_atexit':
atexit_arm.cc:(.text.__aeabi_atexit+0x0): multiple definition of `__aeabi_atexit'
/home/rgilmore/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/softfp/libc_s.a(lib_a-aeabi_atexit.o):aeabi_atexit.c:(.text.__aeabi_atexit+0x0): first defined here
/home/rgilmore/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/softfp/libstdc++_s.a(del_op.o): In function `operator delete(void*)':
del_op.cc:(.text._ZdlPv+0x0): multiple definition of `operator delete(void*)'
/home/rgilmore/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/softfp/libstdc++_s.a(del_op.o):del_op.cc:(.text._ZdlPv+0x0): first defined 

### Expected Results
Should build successfully does not build

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```[env:nucleo_f401re]
platform = ststm32
framework = mbed
board = nucleo_f401re
upload_port = /media/rgilmore/NODE_F401RE/
```

**Source file to reproduce issue:**
```
#include "mbed.h"
#include "Crypto.h"
//------------------------------------
// Hyperterminal configuration
// 115200 bauds, 8-bit data, no parity
//------------------------------------

Serial pc(USBTX, USBRX);

DigitalOut myled(LED1);

int main() {
  int i = 1;
  int j=0;
  pc.baud(115200);
  uint8_t key[16] = { 0x2b, 0x7e, 0x15, 0x16, 0x28, 0xae, 0xd2, 0xa6, 0xab, 0xf7, 0x15, 0x88, 0x09, 0xcf, 0x4f, 0x3c};
  uint8_t plaintext[16] = {0x6b, 0xc1, 0xbe, 0xe2, 0x2e, 0x40, 0x9f, 0x96, 0xe9, 0x3d, 0x7e, 0x11, 0x73, 0x93, 0x17, 0x2a};
  uint8_t ciphertext[16];
  while(1) {
      wait(0.1);
      AES myAES(AES_128,key);
      myAES.encrypt(ciphertext,plaintext,16);
      j=i%16;
      i++;
      pc.printf("%d\n", ciphertext[j]);
      myled = !myled;
  }
}
```

### Additional info

ivankravets, 2016-03-11:
Thanks. Please re-test in the latest http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-03-11 by ivankravets

gilandose, 2016-03-11:
Thanks, for that builds fine now no errors.

commit 757df8c8 (2016-03-11): Fix multiple definition errors for ST STM32 development platform and mbed framework // Resolve #571