platformio/platformio-core #573: Auto-complete does not work for external libraries

ogail, 2016-03-12:
### Configuration

**Operating system**: mac os x el-capitan

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.8.5

### Description of problem
I followed same pattern mentioned in PlatformIO example folders for importing header files from external libs and it the build succeeds with #include <HeaderFile.h> but the problem is when I define object from class that's from HeaderFile the auto-complete thinks it's of type `int`. To overcome this problem I had to include the complete path to the HeaderFile, like
```
#include </Users/elogeela/Documents/projects/zumo-line-follower/lib/ZumoMotors/ZumoMotors.h>
```

#### Steps to Reproduce

1. Create src/main.cpp file with contents provided below
2. Try auto-complete motors. and it will not show anything
3. Update the #include <Motors.h> to have full path
4. Now auto-complete works

### Actual Results
Auto-complete does not work with #include <Motors.h>

### Expected Results
Auto-complete should work with just #include <Motors.h>

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <ZumoMotors.h>

#define LED_PIN 13

ZumoMotors motors;

void setup()
{
  pinMode(LED_PIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_PIN, HIGH);
  motors.setSpeeds(50, 50);
  delay(500);
}
```

### Additional info
The repo with the folder structure is here: https://github.com/ogail/zumo-line-follower

orgkhnargh, 2016-03-12:
This issue seems to be a duplicate of platformio/platformio-atom-ide/issues/65. You need to click "PlatformIO > Rebuild C/C++ Project Index (Autocomplete, Linter)" after changes to the libraries.

ogail, 2016-03-12:
Thanks this fixed the issue!

closed 2016-03-12 by ogail