platformio/platformio-core #575: Lib not found

nlamirault, 2016-03-12:
### Configuration

**Operating system**:

Arch Linux

**PlatformIO Version** (`platformio --version`):

```bash
$ platformio --version
PlatformIO, version 2.8.5
```

### Description of problem

Can't build project due error with missing file headers.

#### Steps to Reproduce

1. Clone project [1], checkout develop branch.
2. Try to build the project : 

```bash
$ . venv/bin/activate
$ platformio lib list
[ ID  ] Name             Compatibility         "Authors": Description
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ 89  ] PubSubClient     arduino, atmelavr, espressif "Nick O'Leary": A client library for MQTT messaging. MQTT is a lightweight messaging protocol ideal for small devices. This library allows you to send and receive MQTT messages. It supports the latest MQTT 3.1.1 protocol and can be configured to use the older MQTT 3.1 if needed. It supports all Arduino Ethernet Client compatible hardware, including the Intel Galileo/Edison, ESP8266 and TI CC3000.
[1101 ] ESP8266wifi      arduino, atmelavr     "Jonas Ekstrand": ESP8266 Arduino library with built in reconnect functionality
```

### Actual Results

```bash
$ platformio run -d arduino/teleinfo
[Sat Mar 12 17:42:28 2016] Processing arduino_uno (platform: atmelavr, lib_install: 89,1101, upload_port: /dev/ttyACM1, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/arduino_uno/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -DPLATFORMIO=020805 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant -I.pioenvs/arduino_uno/PubSubClient_ID89 -I.pioenvs/arduino_uno/SoftwareSerial src/tmp_ino_to.cpp
src/tmp_ino_to.cpp:18:25: fatal error: ESP8266WiFi.h: No such file or directory
#include <ESP8266WiFi.h>
^
compilation terminated.
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial0.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial0.cpp
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial1.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial1.cpp
scons: *** [.pioenvs/arduino_uno/src/tmp_ino_to.o] Error 1
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial2.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial2.cpp
========================================================================== [ ERROR ] Took 0.30 seconds ==========================================================================
Makefile:69: recipe for target 'arduino-build' failed
make: *** [arduino-build] Error 1
```

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
src_dir = src

[env:arduino_uno]
platform = atmelavr
framework = arduino
board = uno
lib_install = 89,1101

# you may need to change this to /dev/ttyACM0 or just comment out
upload_port = /dev/ttyACM1

```

[1] : https://github.com/zeiot/jarvis

ivankravets, 2016-03-12:
Did you push the latest code to CVS? See my output from `develop`
```shell
(develop) ikravets@innomac /V/S/P/G/p/origin> pio run -d /Users/ikravets/Downloads/jarvis-develop/arduino/teleinfo
[Sat Mar 12 18:54:51 2016] Processing arduino_uno (platform: atmelavr, lib_install: 89,1101, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
The libraries with IDs '89, 1101' have not been installed yet. Would you like to install them now? [y/N]: y
Installing library [ 89 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #89 'PubSubClient' has been successfully installed!
Installing library [ 1101 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #1101 'ESP8266wifi' has been successfully installed!
avr-g++ -o .pioenvs/arduino_uno/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -DPLATFORMIO=020806 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant -I.pioenvs/arduino_uno/ESP8266wifi_ID1101 -I.pioenvs/arduino_uno/PubSubClient_ID89 -I.pioenvs/arduino_uno/SoftwareSerial src/tmp_ino_to.cpp
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial1.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial1.cpp
jarvis-teleinfo.ino:31:1: error: 'WiFiClient' does not name a type
jarvis-teleinfo.ino: In function 'void setup_wifi()':
jarvis-teleinfo.ino:41:8: error: 'class HardwareSerial' has no member named 'printf'
jarvis-teleinfo.ino:43:1: error: 'WiFi' was not declared in this scope
jarvis-teleinfo.ino:44:25: error: 'WL_CONNECTED' was not declared in this scope
jarvis-teleinfo.ino:49:8: error: 'class HardwareSerial' has no member named 'printf'
jarvis-teleinfo.ino: In function 'void callback(char*, byte*, unsigned int)':
jarvis-teleinfo.ino:55:5: error: 'i' was not declared in this scope
jarvis-teleinfo.ino:56:1: error: 'message_buff' was not declared in this scope
jarvis-teleinfo.ino:58:1: error: 'message_buff' was not declared in this scope
jarvis-teleinfo.ino:58:14: error: 'i' was not declared in this scope
jarvis-teleinfo.ino: In function 'void setup_mqtt()':
jarvis-teleinfo.ino:81:61: error: 'wifiClient' was not declared in this scope
jarvis-teleinfo.ino: In function 'void loop()':
jarvis-teleinfo.ino:101:1: error: 'client' was not declared in this scope
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial2.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial2.cpp
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial3.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial3.cpp
scons: *** [.pioenvs/arduino_uno/src/tmp_ino_to.o] Error 1
================================================================= [ ERROR ] Took 4.03 seconds =================================================================
(develop) ikravets@innomac /V/S/P/G/p/origin>
```

nlamirault, 2016-03-12:
Checkout project : 

```bash
[nlamirault]:~$ cd /tmp/
[nlamirault]:/tmp$ git clone https://github.com/zeiot/jarvis
Cloning into 'jarvis'...
remote: Counting objects: 1225, done.
remote: Compressing objects: 100% (910/910), done.
remote: Total 1225 (delta 309), reused 1191 (delta 275), pack-reused 0
Receiving objects: 100% (1225/1225), 6.76 MiB | 3.93 MiB/s, done.
Resolving deltas: 100% (309/309), done.
Checking connectivity... done.
[nlamirault]:/tmp$ cd jarvis/
[nlamirault]:/tmp/jarvis [master*]$ ls
CONTRIBUTING.md  LICENSE  README.md  raspbian.sh
[nlamirault]:/tmp/jarvis [master*]$ git checkout develop 
Branch develop set up to track remote branch develop from origin.
Switched to a new branch 'develop'
```

Create a virtualenv and install platformio : 

```bash
[nlamirault]:/tmp/jarvis [develop*]$ virtualenv --python=/usr/bin/python2 venv 
Running virtualenv with interpreter /usr/bin/python2
New python executable in /tmp/jarvis/venv/bin/python2
Also creating executable in /tmp/jarvis/venv/bin/python
Installing setuptools, pip, wheel...done.
[nlamirault]:/tmp/jarvis [develop*]$ . venv/bin/activate
(venv) [nlamirault]:/tmp/jarvis [develop*]$ pip install platformio
Collecting platformio
  Using cached platformio-2.8.5-py27-none-any.whl
Collecting click<6,>=3.2 (from platformio)
  Using cached click-5.1-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio)
  Using cached requests-2.9.1-py2.py3-none-any.whl
Collecting pyserial<4 (from platformio)
Collecting colorama (from platformio)
  Using cached colorama-0.3.7-py2.py3-none-any.whl
Collecting bottle<0.13 (from platformio)
Collecting lockfile<0.13,>=0.9.1 (from platformio)
  Using cached lockfile-0.12.2-py2.py3-none-any.whl
Installing collected packages: click, requests, pyserial, colorama, bottle, lockfile, platformio
Successfully installed bottle-0.12.9 click-5.1 colorama-0.3.7 lockfile-0.12.2 platformio-2.8.5 pyserial-3.0.1 requests-2.9.1
```

Run : 

```bash
(venv) [nlamirault]:/tmp/jarvis [develop*]$ pio run -d arduino/teleinfo
[Sat Mar 12 18:36:19 2016] Processing arduino_uno (platform: atmelavr, upload_port: /dev/ttyACM1, board: uno, lib_use: PubSubClient, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/arduino_uno/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CP
U=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -DPLATFORMIO=020805 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant -I.pi
oenvs/arduino_uno/PubSubClient_ID89 -I.pioenvs/arduino_uno/SoftwareSerial src/tmp_ino_to.cpp
avr-ar rcs .pioenvs/arduino_uno/libFrameworkArduinoVariant.a
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/CDC.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p 
-DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_un
o/FrameworkArduino/CDC.cpp
avr-ranlib .pioenvs/arduino_uno/libFrameworkArduinoVariant.a
src/tmp_ino_to.cpp:18:25: fatal error: ESP8266WiFi.h: No such file or directory
#include <ESP8266WiFi.h>
^
compilation terminated.
scons: *** [.pioenvs/arduino_uno/src/tmp_ino_to.o] Error 1
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/HardwareSerial.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=
atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs
/arduino_uno/FrameworkArduino/HardwareSerial.cpp
========================================================================== [ ERROR ] Took 0.32 seconds ==========================================================================
```

Then i try to use lib_install parameter for dependencies in platformio.ini file : 

```ini
lib_install = 89,1101
```
Strange, i've got only one dependency in the pioenvs directory : 

```bash
$ ls -1 arduino/teleinfo/.pioenvs/arduino_uno/
FrameworkArduino
FrameworkArduinoVariant
PubSubClient_ID89
SoftwareSerial
libFrameworkArduinoVariant.a
src
```

ivankravets, 2016-03-12:
Give me please an output of 
```
(venv) [nlamirault]:/tmp/jarvis [develop*]$ pio lib list
```

I don't know where do you run it but it looks that libraries are not installed. Please specify `lib_install = 89,1101` in platformio.ini. I don't see here `[Sat Mar 12 18:36:19 2016] Processing arduino_uno (platform: atmelavr, upload_port: /dev/ttyACM1, board: uno, lib_use: PubSubClient, framework: arduino)` it.

nlamirault, 2016-03-12:
```bash
$ pio lib list
[ ID  ] Name             Compatibility         "Authors": Description
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ 89  ] PubSubClient     arduino, atmelavr, espressif "Nick O'Leary": A client library for MQTT messaging. MQTT is a lightweight messaging protocol ideal for small devices. This library allows you to send and receive MQTT messages. It supports the latest MQTT 3.1.1 protocol and can be configured to use the older MQTT 3.1 if needed. It supports all Arduino Ethernet Client compatible hardware, including the Intel Galileo/Edison, ESP8266 and TI CC3000.
[1101 ] ESP8266wifi      arduino, atmelavr     "Jonas Ekstrand": ESP8266 Arduino library with built in reconnect functionality
```
```bash
$ find . -name "platformio.ini"
./arduino/platformio.ini
./arduino/teleinfo/platformio.ini
./arduino/dht/platformio.ini
```
```bash
 $ ls -l ./arduino/teleinfo/platformio.ini ./arduino/dht/platformio.ini ./arduino/platformio.ini
lrwxrwxrwx 1 nlamirault nlamirault  17 Mar 12 18:35 ./arduino/dht/platformio.ini -> ../platformio.ini
-rw-r--r-- 1 nlamirault nlamirault 368 Mar 12 18:44 ./arduino/platformio.ini
lrwxrwxrwx 1 nlamirault nlamirault  17 Mar 12 18:35 ./arduino/teleinfo/platformio.ini -> ../platformio.ini

```

ivankravets, 2016-03-12:
Hm.. Please
```
(venv) [nlamirault]:/tmp/jarvis [develop*]$ pio run -d arduino/teleinfo -t idedata
```

nlamirault, 2016-03-12:
```bash
$ pio run -d arduino/teleinfo -t idedata
[Sat Mar 12 19:01:19 2016] Processing arduino_uno (platform: atmelavr, lib_install: 89,1101, upload_port: /dev/ttyACM1, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
{"defines": ["F_CPU=16000000L", "ARDUINO_ARCH_AVR", "ARDUINO_AVR_UNO", "ARDUINO=10607", "PLATFORMIO=020805", "__AVR_ATmega328P__"], "cc_flags": "-std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -DPLATFORMIO=020805", "cxx_path": "/home/nlamirault/.platformio/packages/toolchain-atmelavr/bin/avr-g++", "cxx_flags": "-fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -DPLATFORMIO=020805", "includes": ["/home/nlamirault/.platformio/packages/framework-arduinoavr/variants/standard", "/home/nlamirault/.platformio/packages/framework-arduinoavr/cores/arduino", "/home/nlamirault/.platformio/lib/PubSubClient_ID89/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/SoftwareSerial", "/home/nlamirault/.platformio/lib/ESP8266wifi_ID1101", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Audio/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Bridge/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/EEPROM", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Esplora/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Ethernet/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Firmata", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/GSM/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/HID", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Keyboard/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/LiquidCrystal/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Mouse/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/RobotIRremote/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Robot_Control/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Robot_Motor/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/SD/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/SPI", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Scheduler/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Servo/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/SpacebrewYun/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Stepper/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/TFT/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Temboo/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/USBHost/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/WiFi/src", "/home/nlamirault/.platformio/packages/framework-arduinoavr/libraries/Wire", "/home/nlamirault/.platformio/packages/toolchain-atmelavr/avr/include", "/home/nlamirault/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include-fixed", "/home/nlamirault/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include"]}
========================================================================== [SUCCESS] Took 0.21 seconds ==========================================================================
```

In $HOME/.platformio directory i see libs : 

```bash
$ ls ~/.platformio/lib/
ESP8266wifi_ID1101  PubSubClient_ID89

```

```bash
$ rm -fr ~/.platformio/
$ pio run -d arduino/teleinfo

*********************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/#!/platformio-ide
- donate to keep PlatformIO alive! > http://platformio.org/#!/donate
*********************************************************************************************************************************************************************************

[Sat Mar 12 19:02:50 2016] Processing arduino_uno (platform: atmelavr, lib_install: 89,1101, upload_port: /dev/ttyACM1, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
The libraries with IDs '89, 1101' have not been installed yet. Would you like to install them now? [y/N]: y
Installing library [ 89 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #89 'PubSubClient' has been successfully installed!
Installing library [ 1101 ]:
[... cut ...]
avr-g++ -o .pioenvs/arduino_uno/FrameworkArduino/PluggableUSB.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/arduino_uno/FrameworkArduino -I.pioenvs/arduino_uno/FrameworkArduinoVariant .pioenvs/arduino_uno/FrameworkArduino/PluggableUSB.cpp
========================================================================= [ ERROR ] Took 50.12 seconds =========================================================================
```

ivankravets, 2016-03-12:
I got it. Compiler is right. You OS is case sensitive. There is no `ESP8266WiFi.h` file.

closed 2016-03-12 by ivankravets

nlamirault, 2016-03-12:
Thanks for your help ! Sorry for noise

ivankravets, 2016-03-12:
Please give us a star :star: https://github.com/platformio/platformio/stargazers

Thanks.