platformio/platformio-core #577: upload error, while loading shared libraries: libusb-1.0.so.0

matak, 2016-03-14:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums http://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [x] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

[root@foceni photorobot-arduino]# cat /etc/*release*
CentOS release 6.7 (Final)
LSB_VERSION=base-4.0-amd64:base-4.0-noarch:core-4.0-amd64:core-4.0-noarch
cat: /etc/lsb-release.d: Is a directory
CentOS release 6.7 (Final)
CentOS release 6.7 (Final)
cpe:/o:centos:linux:6:GA

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 2.8.5

### Description of problem

when i try to upload compiled script then i get this
```

[Mon Mar 14 03:16:53 2016] Processing megaatmega2560 (platform: atmelavr, board: megaatmega2560, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/megaatmega2560/firmware.hex"])
Auto-detected UPLOAD_PORT/DISK: /dev/ttyACM0
"/root/.platformio/packages/tool-avrdude/avrdude" -v -p atmega2560 -C "/root/.platformio/packages/tool-avrdude/avrdude.conf" -c wiring -b 115200 -P /dev/ttyACM0 -D -U flash:w:.pioenvs/megaatmega2560/firmware.hex:i
/root/.platformio/packages/tool-avrdude/avrdude: error while loading shared libraries: libusb-1.0.so.0: cannot open shared object file: No such file or directory
scons: *** [upload] Error 127
========================================================== [ ERROR ] Took 1.26 seconds ==========================================================
[root@foceni photorobot-arduino]#

```

matak, 2016-03-14:
the solution is 

 yum install libusb1

closed 2016-03-14 by matak

ivankravets, 2016-03-14:
Thanks a lot, I've updated your https://github.com/platformio/platformio/issues/577#issue-140558445