platformio/platformio-core #578: Error: The package 'toolchain-gccarmlinuxgnueabi' is not available for your system 'linux_x86_64'

wintersandroid, 2016-03-14:
Attempt to use the arm / raspberrypi platform. gets the error.
Error: The package 'toolchain-gccarmlinuxgnueabi' is not available for your system 'linux_x86_64'

Also had this error under 64 bit windows.

------------------------------------------------------------------

### Configuration

**Operating system**: Linux, x86_64 x-ubuntu

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.8.5

### Description of problem

#### Steps to Reproduce

1. Using the IDE, Initalise new project
2. Select "Raspberry Pi 1 Model B"
3. Click Initalize

### Actual Results
PlaftormIO: Failed to initialize PlatformIO project!
Error: The package 'toolchain-gccarmlinuxgnueabi' is not available for your system 'linux_x86_64'

### Expected Results
I would expect to be able to cross compile to arm from 64 bit linux..

ivankravets, 2016-03-14:
Are you going to build program for desktop `linux_x86_64` using `Raspberry Pi`?

ivankravets, 2016-03-14:
@wintersandroid could you help us with these toolchains? Because we have only for Mac OS X. See https://bintray.com/platformio/dl-packages/toolchain-gccarmlinuxgnueabi/view

wintersandroid, 2016-03-14:
I can have a look at it.. :)

JohnnySheppard, 2016-04-27:
I'm afraid I don't know much about cross compiling and toolchains yet, but I've been looking into it a little. Does this link help anyone?: https://github.com/raspberrypi/tools

ejb1123, 2016-05-14:
yes that is the link that you get the cross compiler from

ivankravets, 2016-11-27:
Moved to https://github.com/platformio/platform-linux_arm/issues/2

closed 2016-11-27 by ivankravets