platformio/platformio-core #580: Reports errors that do not exist

charles-owen, 2016-03-15:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums http://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [X ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

Windows 10

IDE 1.0.5, CLI 2.8.5 (`platformio --version`):

### Description of problem

When I compile a program the IDE reports errors in the source code that are not emitted by the compiler in the build window. I am loading main.c from stm32-spl-blink. It compiles and runs just fine. But, the IDE says I have two errors where none exist. 

#### Steps to Reproduce

1. New project for disco_l152rb with platformio.ini as below.
2. Set main.c to the source code below.
3. Build. 

### Actual Results

arm-none-eabi-gcc -o .pioenvs\disco_l152rb\src\main.o -c -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DF_CPU=32000000L -DST
M32L152XB -DSTM32L1 -DSTM32L152xB -DSTM32L1XX_MD -DPLATFORMIO=020805 -I.pioenvs\disco_l152rb\FrameworkCMSIS -I.pioenvs\disco_l152rb\FrameworkCMSISVariant -I.pioen
vs\disco_l152rb\FrameworkSPLInc -I.pioenvs\disco_l152rb\FrameworkSPL src\main.c

No error is indicated. But, the IDE indicates:

GCC error variable or field 'simple_delay' declared void at line 22 col 19
GCC error 'uint32_t' was not declared in this scope at line 22 col 19

### Expected Results

There should not be any error messages.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:disco_l152rb]
platform = ststm32
framework = spl
board = disco_l152rb
```

**Source file to reproduce issue:**
```cpp
#ifdef STM32L1
	#include <stm32l1xx_gpio.h>
	#include <stm32l1xx_rcc.h>
	#define LEDPORT (GPIOB)
	#define LEDPIN (GPIO_Pin_7)
	#define ENABLE_GPIO_CLOCK (RCC_AHBPeriphClockCmd(RCC_AHBPeriph_GPIOB, ENABLE))
#elif STM32F3
	#include <stm32f30x_gpio.h>
	#include <stm32f30x_rcc.h>
	#define LEDPORT (GPIOE)
	#define LEDPIN (GPIO_Pin_8)
	#define ENABLE_GPIO_CLOCK (RCC_AHBPeriphClockCmd(RCC_AHBPeriph_GPIOE, ENABLE))
#elif STM32F4
	#include <stm32f4xx_gpio.h>
	#include <stm32f4xx_rcc.h>
	#define LEDPORT (GPIOD)
	#define LEDPIN (GPIO_Pin_12)
	#define ENABLE_GPIO_CLOCK (RCC_AHB1PeriphClockCmd(RCC_AHB1Periph_GPIOD, ENABLE))
#endif

/* timing is not guaranteed :) */
void simple_delay(uint32_t us)
{
	/* simple delay loop */
	while (us--) {
		asm volatile ("nop");
	}
}

/* system entry point */
int main(void)
{
	/* gpio init struct */
	GPIO_InitTypeDef gpio;
	/* reset rcc */
	RCC_DeInit();
	/* enable clock GPIO */
	ENABLE_GPIO_CLOCK;
	/* use LED pin */
	gpio.GPIO_Pin = LEDPIN;
	/* mode: output */
	gpio.GPIO_Mode = GPIO_Mode_OUT;
	/* output type: push-pull */
	gpio.GPIO_OType = GPIO_OType_PP;
	/* apply configuration */
	GPIO_Init(LEDPORT, &gpio);
	/* main program loop */
	for (;;) {
		/* set led on */
		GPIO_SetBits(LEDPORT, LEDPIN);
		/* delay */
		simple_delay(100000);
		/* clear led */
		GPIO_ResetBits(LEDPORT, LEDPIN);
		/* delay */
		simple_delay(100000);
	}

	/* never reached */
	return 0;
}

```

### Additional info

gerritv, 2016-03-15:
Please see Issue 543: https://github.com/platformio/platformio/issues/543 What you are seeing is Intellisense errors, not compile errors. The solution is a bit finicky depending on your source code layout.

ivankravets, 2016-03-15:
How did you create project? Try `Menu: PlatformIO > Rebuild C/C++ Project Index`.

charles-owen, 2016-03-15:
I built the project using the instructions in the getting started, just changed the board type.

But, Rebuild C/C++ Project Index does appear to have fixed the problem. What does that do?

ivankravets, 2016-03-15:
@charles-owen you have changed the board manually without rebuilding index. Code Linter didn't know about it. I've just opened an issue for that https://github.com/platformio/platformio-atom-ide/issues/70

closed 2016-03-15 by ivankravets