platformio/platformio-core #585: MightyCore Boards Support

peekpt, 2016-03-18:
https://github.com/MCUdude/MightyCore

Please consider to add support for this boards which supports:

-ATmega1284, ATmega644, ATmega324, ATmega164, ATmega32, ATmega16 and ATmega8535
-Optiboot 6.0
-Multiple voltages and freqs.

Please consider to add this bunch of boards from MightyCore to your database, since the support for ATmega644/1284 from other boards are like sanguino, microduino are not updated anymore.

Thank you!

ivankravets, 2016-04-28:
@peekpt @MCUdude , please re-test with the latest http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-04-28 by ivankravets

MCUdude, 2016-05-06:
I've just started using the PlatformIO IDE (Atom), and just discovered that there are noe BOD option or clock frequency option available for the MightyCore compatible microcontrollers. Is it possible to add this in a separate menu, like it it on the Arduino IDE, or does it have to be like:
ATmega1284 @ 20MHz external
ATmega1284 @ 16MHz external
ATmega1284 @ 8MHz external
.. and so on.

Oh, I almost forgot. You forgot to add the ATmega8535 to the list 😉

ivankravets, 2016-05-06:
@MCUdude You can specify own frequency. See http://docs.platformio.org/en/latest/projectconf.html#board-f-cpu

peekpt, 2016-05-06:
I think you can add a Json configuration to your git project

ivankravets, 2016-05-06:
> Oh, I almost forgot. You forgot to add the ATmega8535 to the list 

@MCUdude do you have this board?

MCUdude, 2016-05-06:
> @MCUdude do you have this board?
Yes I do. The ATmega8535 is the smallest (8kB, 512b of RAM) microcontroller that MightyCore supports

peekpt, 2016-05-06:
There's a LOT of combinations with this board.
I think there as to be at least a board with 5v and 3.3v for each processor because of the fuses. The rest can be configured in platform.ini

**Supported** Voltages:
3.3v and 5v
**Supported** microcontrollers
ATmega1284*
ATmega644*
ATmega324*
ATmega164*
ATmega32*
ATmega16*
ATmega8535*
**Supported** clock frequencies
20 MHz external oscillator
16 MHz external oscillator (default)
8 MHz external oscillator
8 MHz internal oscillator
1 MHz internal oscillator

MCUdude, 2016-05-06:
I don't think a "Supported voltages" menu is the best way to do it. Some of the low power ones can run down to 1.8v @ 1MHz, but it doesn't mean you have to. I suggest a BOD (Brown out detection) menu and a clock menu

ivankravets, 2016-05-06:
@peekpt 
> I think there as to be at least a board with 5v and 3.3v for each processor because of the fuses. 

We don't use fuses. They are used only to flash bootloader (where PlatformIO doesn't allow it)

> Yes I do. The ATmega8535 is the smallest (8kB, 512b of RAM) microcontroller that MightyCore supports

We will add support for this board.

MCUdude, 2016-05-06:
> We don't use fuses. They are used only to flash bootloader (where PlatformIO doesn't allow it)

OK, so it's not possible to burn the fuses in PlatformIO Atom? Hmm, this might be an issue. When you buy a brand new AVR microcontroller, the 1MHz internal oscillator is enabled by default, and the fuse settings need to be changed to run with an external oscillator. That means you can't rely only on PlatformIO, you'll have to install Arduino IDE or Avrdude to set the fuses and burn the bootloader (if I've understood correctly)

ivankravets, 2016-05-06:
@MCUdude PlatformIO contains `avrdude` binary and uses it to upload firmware. The user has the ability to create an own target and to specify the command with fuses. This is the first occurrence where these fuses are needed.

See https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/atmelavr.py#L113

ivankravets, 2016-05-06:
The ATmega8535 has been added. Please test it with  http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2016-05-06:
@MCUdude do you need any help with fuses? You need [extra_script](http://docs.platformio.org/en/latest/projectconf.html#extra-script) and additional target.

peekpt, 2016-05-06:
Well I'm on mac and use AVRFuses,it's a great little tool. @MCUdude platformio is great and simple and easy to use, please don't give up on it. I tested for a while various eclipse plugins and I find Platformio  to be the most lightweight and productive and you can add plugins to it. Code linter works well to. It has the ability to work on eclipse and vim and the best it's has a bash interface. With CLI you can code on a shell with vim nano, emacs, without the worry of compiling commands and updating libs and plats and boards. You can add plugins for git, for symbols etc...

MCUdude, 2016-05-06:
> @MCUdude do you need any help with fuses? You need extra_script and additional target.

Oh, this is starting to get heavy! I know very well how the Arduino IDE handles fuses, serial ports and bootloader burning, but I have no experience with PlatformIO. Still I feel the urge for migrating for something more professional; an IDE that actually helps you while writing code.

I was hoping that there was possible to just select your preferred clock frequency and BOD option in the GUI, and then set the fuses using your preferred programmer. If this was an option I see no reason why (Arduino) beginners shouldn't start 'using PlatformIO right away.

@peekpt I'm on mac as well. I'm comfortable using the command line, but I'm not "a make file person", and prefer to do my work in an IDE. I've been using Arduino IDE for the past years, but It feels very limiting, so I've been using an external text editor (BBEdit) instead, and just did the uploading in Arduino IDE.

peekpt, 2016-05-06:
once everything is configured properly you still could use BBEdit and type "pio run" to compile and upload, easy as that.

MCUdude, 2016-05-06:
I'm happy for any help I can get. I want to create a simple guide on my Github repo for using the MightyCore with PlatformIO for those who have never heard of PlatformIO. Is there anything I can add to my repository to make the setup process even simpler (preconfigured files etc.)? I think this will lower the bar significantly for the ones that feel like the Arduino IDE is too basic.

What I know so far:
* How to install (I'm using a mac, but installation should be trivial)
* How to set up your project folder
* How to select your board/micro controller

Loose ends:
* How to change the clock frequency fuses (preferably in GUI or simple commands)
* How to change the BOD setting (preferably in GUI or simple commands)
* How to burn bootloader (preferably in GUI or simple commands)
* Upload using programmer
* Select the serial port for uploading (is this done in the platform.ini file or in the GUI?)

peekpt, 2016-05-06:
> Upload using programmer:

_You can set various targets in platform.ini_

    #645 with bootloader
    [env:sparkfun_promicro8]
    platform = atmelavr
    framework = arduino
    board = sparkfun_promicro8

    #with usbasp via ISP
    [env:sparkfun_promicro8]
    platform = atmelavr
    framework = arduino
    board = sparkfun_promicro8
    upload_protocol = usbasp

  did you check this sublimetext integration?  

http://docs.platformio.org/en/latest/ide/sublimetext.html#integration

peekpt, 2016-05-06:
>Select the serial port for uploading

It will automatically find the serial port for you but you can specify on platformio.ini
you can use:

     pio serialposrts list

MCUdude, 2016-05-06:
> You can set various targets in platform.ini

# with bootloader
[env:sparkfun_promicro8]
platform = atmelavr
framework = arduino
board = sparkfun_promicro8

#with usbasp via ISP
[env:sparkfun_promicro8]
platform = atmelavr
framework = arduino
board = sparkfun_promicro8
upload_protocol = usbasp

Thanks! That makes sense 😃 

Is it possible for the IDE to automatically add the fields:
```
#Clock speed:   Baud rate:
#20000000       115200
#16000000       115200
#8000000         57600
#1000000         9600
  
board_f_cpu = 16000000L
upload_speed = 115200
```

to the platformio.ini file when using one of the MightyCore supported microcontrollers. If the user are going to change the clock frequency the upload speed needs to be changed too.

ivankravets, 2016-05-06:
@MCUdude I recommend starting from PlatformIO CLI. PlatformIO IDE, Eclipse, Deviot Plugin for Sublime Text and etc., they use PlatformIO CLI in the core. Later, we will discuss IDEs.

Let's start from the scratch... From the PlatformIO CLI...

1. Install PlatformIO http://docs.platformio.org/en/latest/installation.html
2. Take a look at [Quick Start](http://docs.platformio.org/en/latest/quickstart.html)
3. Create new directory. Open system Terminal and change directory to it.
4. `pio init --board mightycore1284` or `pio init -b mightycore1284`
5. Place code to `src` folder
6. Build project `pio run`
7. Upload firmware `pio run --target upload` or `pio run -t upload`.

ivankravets, 2016-05-06:
@MCUdude 

Loose ends:

> How to change the clock frequency fuses (preferably in GUI or simple commands)

http://docs.platformio.org/en/latest/projectconf.html#board-f-cpu

> How to change the BOD setting (preferably in GUI or simple commands)

We don't have option for that. Nevertheless, user has ability to create own board preset and specify own settings (`build > variant`). See http://docs.platformio.org/en/latest/platforms/creating_board.html

Just copy one of existing data from https://github.com/platformio/platformio/blob/develop/platformio/boards/mightycore.json

> How to burn bootloader (preferably in GUI or simple commands)

Tell me please the full `avrdude` command and I will create target for you.

> Upload using programmer

http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

> Select the serial port for uploading (is this done in the platform.ini file or in the GUI?)

* http://docs.platformio.org/en/latest/projectconf.html#upload-port
* `pio run -t upload --upload-port /dev/null`

P.S: The problem here is that we don't support only Arduino or AVR based boards. PlatformIO supports 250+ different embedded boards. We can't create GUI for all features from hundreds boards. In this case, we allow users to customize build process using `platformio.ini`. Of course, we are interested in simple process for the all boards but it isn't easy. We are working on it.

MCUdude, 2016-05-06:
> Let's start from the scratch... From the PlatformIO CLI...

Install PlatformIO http://docs.platformio.org/en/latest/installation.html
Take a look at Quick Start
Create new directory. Open system Terminal and change directory to it.
pio init --board mightycore1284 or pio init -b mightycore1284
Place code to src folder
Build project pio run
Upload firmware pio run --target upload or pio run -t upload. If you need to upload firmware using programmer, use pio run -t program instead. See details.

Wow, this stuff is REALLY great! :smiley: thanks a lot!

> Tell me please the full avrdude command and I will create target for you.

I don't know what a target would look like in PlatformIO, nor how to use one, but if you explain it, I'll be really grateful! There are several quirks related to the bootloader burning. First AVRdude set the correct fuses with one command, then the bootloader is uploaded using another command. I'm away for the weekend, so I'm not able to recreate the exact AVRdude commands. But let's pretend that the commands below are correct.
Here's the fuse setting command for setting used with the 16MHz bootloader (and BOD disabled):
`avrdude -C ~/Documents/Arduino/hardware/MightyCore/avr/avrdude.conf -v -p atmega1284p -c usbtiny -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xde:m -Ulfuse:w:0xf7:m 
`

After the fuses are set, AVRdude uploads the bootloader:
`
avrdude -C ~/Documents/Arduino/hardware/MightyCore/avr/avrdude.conf -v -p atmega1284p -c usbtiny -e -Uflash:w:~/Documents/Arduino/hardware/MightyCore/avr/bootloaders/optiboot/atmega1284/optiboot_atmega1284p_16MHz.hex:i -Ulock:w:0x0f:m
`

The problem is that many of the same microcontrollers (e.g ATmega1284 and ATmega1284p) got different device signatures, and therefore needs a different part number. How do we deal with that? I created another menu option for the affected microcontrollers.

ivankravets, 2016-05-07:
@MCUdude 

# `platformio.ini`
```ini
[env:mightycore1284_set_fuses]
platform = atmelavr
framework = arduino
board = mightycore1284
upload_protocol = usbtiny
extra_script = extra_script.py
upload_flags = -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xde:m -Ulfuse:w:0xf7:m

[env:mightycore1284_upload_bootloader]
platform = atmelavr
framework = arduino
board = mightycore1284
upload_protocol = usbtiny
extra_script = extra_script.py
upload_flags = -e -Uflash:w:~/Documents/Arduino/hardware/MightyCore/avr/bootloaders/optiboot/atmega1284/optiboot_atmega1284p_16MHz.hex:i -Ulock:w:0x0f:m
```
# `extra_script.py`

Please the code below to `extra_script.py` file and this file should be located near `platformio.ini` file.

```python
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(PROGRAMHEXCMD='"$UPLOADER" $UPLOADERFLAGS')
```

-----

1. Set fuses = `pio run -t program -e mightycore1284_set_fuses`
2. Upload bootloader = `pio run -t program -e mightycore1284_upload_bootloader`

peekpt, 2016-05-11:
@MCUdude I finally received my chips atmega644pa for a personal project. I'm using your mighty core standard mapping. I leave my config. So far I2c works, SPI works. But I had to burn boot loader on Arduino App, used BOD 2.7v for 3.3v vcc. I used usbasp via ISP method, no problem.

    [env:mightycore644]
    platform = atmelavr
    framework = arduino
    board = mightycore644
    board_f_cpu = 8000000L
    upload_speed = 57600

I am running  8MHz@3v3, I could not connect more than 57600 speed. I am using a cp2102 chip.

commit 4f0b4817 (2016-04-28): Add support for MightyCore boards // Resolve #585