platformio/platformio-core #587: avrdude binary buggy

pfrankw, 2016-03-19:
### Configuration

**Operating system**: Ubuntu 14.04

**PlatformIO Version**: 2.8.5

### Description of problem
The avrdude binary seems buggy because it cannot often upload the same code the arduino IDE uploads perfectly.

This avrdude binary version just works randomly, most of the times it goes on error and very rarely works. We can say it works about 10% of times.
Of course the code is not modified, I just repeatedly clicked the upload button until it works.

### Workaround
Just substitute the PlatformIO's avrdude binary with the Arduino IDE's one.

### Configuration
Ubuntu 14.04 VM with Arduino Nano v3 clone with CH340G chip.

ivankravets, 2016-03-19:
@valeros please update `avrdude` for linux host OS.

valeros, 2016-03-19:
@pfrankw Could you provide upload log with this error, please?

pfrankw, 2016-03-19:
Yes, this is a build log with the default avrdude binary:

> avrdude: Device signature = 0x1e950f
> avrdude: reading input file ".pioenvs/nanoatmega328/firmware.hex"
> 
> avrdude: writing flash (3704 bytes):
> Writing | ################################################## | 100% 1.34s
>  
>  
> avrdude: 3704 bytes of flash written
> avrdude: verifying flash memory against .pioenvs/nanoatmega328/firmware.hex:
> avrdude: load data flash data from input file .pioenvs/nanoatmega328/firmware.hex:
> avrdude: input file .pioenvs/nanoatmega328/firmware.hex contains 3704 bytes
> avrdude: reading on-chip flash data:
> Reading | ################################################avrdude: stk500_recv(): programmer is not responding
> scons: *** [upload] Error 1
> ========================= [ ERROR ] Took 9.83 seconds =========================
>

pfrankw, 2016-03-19:
I have some news:

When I manually execute (form the terminal plugin in the Platformio IDE): `platformio run --target upload` I successfully upload the code 100% of times even with the avrdude binary shipped with the project.

But when I use the upload button on the left of the IDE I do get the aforementioned 90% "fail rate".

I understand that this is super-strange and maybe it is related to my configuration (VM + clone arduino) so it is ok if the issue gets a lower priority.

ivankravets, 2016-03-19:
Try please these 2 other ways to upload http://docs.platformio.org/en/latest/ide/atom.html#process-project

1. `Menu: PlatformIO > Upload`
2. `PlatformIO: Upload` (status bar)

pfrankw, 2016-03-19:
I tried them now and they do not work.
I suspect that is something correlated to the "build" plugin, maybe some environment variables that this plugin sets before executing avrdude.

ivankravets, 2016-03-19:
@pfrankw could you provide TeamViewer session for debug? me@ikravets.com

closed 2016-05-04 by ivankravets