platformio/platformio-core #589: Arduino code compiles and uploads with success yet an error message is being thrown.

mettinoktay, 2016-03-20:
An interesting thing is going on. There is no compilation or upload error, but this error I keep getting. Any ideas, solutions maybe? It's disturbing.

[image]

orgkhnargh, 2016-03-20:
Click `PlatformIO > Rebuild C/C++ Project Index (Autocomplete, Linter)`.

mettinoktay, 2016-03-20:
Solved.

ivankravets, 2016-03-20:
This issue is fixed in the PlatformIO IDE 1.1.0 what will be released tomorrow.

closed 2016-03-20 by ivankravets

cinderblock, 2017-01-15:
This seems to be happening to me.

Project builds and uploads fine but linter is giving me a bunch of `undefined reference` errors.

I think a recent-ish change in Platform.io IDE may have made this happen again as I don't remember seeing these errors last time I worked on my project, about a month or two ago. I could be remembering this wrong.

The linter errors don't seem to be immediately fixed by rebuilding the project index. They do seem to go away after a restart of the editor but come back after any changes.

IDE 1.7.1
Core 3.2.1
Atom 1.13.0

ivankravets, 2017-01-19:
@cinderblock could you share here your `platformio.ini`?

cinderblock, 2017-01-21:
```
[env:nanoatmega328]
platform=atmelavr
board=nanoatmega328
framework=arduino
lib_deps=
  NeoPixelBus

[platformio]
src_dir=Angle_read_0_1
```

FYI, this is on Windows 10.

I had to add `avr-gcc`'s full path to `linter-gcc`'s settings at some point. Not sure if this was done before we added the `platformio-ide` extension to atom.

This is not happening on a different machine of mine but I haven't gotten the linter setup on there so maybe that's why it's not happening there just yet. (I can't easily get that working until `linter-gcc` fixes the major bug introduced in `v0.7.0`)