platformio/platformio-core #591: Error installing OneWire library [id=1]

nogurenn, 2016-03-22:
### Configuration

**Operating system**: Mac OS X El Capitan (10.11)

**PlatformIO Version**: 2.8.5

### Description of problem
I'm trying to install the OneWire (id: 1) library, but I keep getting this unknown error.
All commands were done within the Atom PlatformIO CLI

#### Steps to Reproduce

1. Open PlatformIO Library Manager CLI
2. Enter `pio lib install 1` (OneWire library)

### Actual Results
```
Installing library [ 1 ]:
Error: Traceback (most recent call last):
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/__main__.py", line 107
, in main
    cli(None, None, None)

  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/core.py", line 700, in __ca
ll__
    return self.main(*args, **kwargs)
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/core.py", line 1027, in inv
oke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/core.py", line 1027, in inv
oke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/core.py", line 873, in invo
ke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/core.py", line 508, in invo
ke
    return callback(*args, **kwargs)
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in
 new_func
    return f(get_current_context(), *args, **kwargs)
File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in
 new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/commands/lib.py", line
 122, in lib_install
    if not lm.install(id_, version):
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/libmanager.py", line 9
5, in install
    dlpath = self.download(dlinfo['url'], gettempdir())
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/libmanager.py", line 3
8, in download
    fd.start()
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/downloader.py", line 6
4, in start
    chunks = int(ceil(self.get_size() / float(self.CHUNK_SIZE)))
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/downloader.py", line 5
9, in get_size
    return int(self._request.headers['content-length'])
  File "/Users/solitude/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/requests/structures.py", line 54,
 in __getitem__
    return self._store[key.lower()][1]
KeyError: 'content-length'
```
### Expected Results
```
Installing library [ 1 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #1 'OneWire' has been successfully installed!
```
### Additional Info
`pip install -U platformio`

```
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/site-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py", line 579, in
setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```

ivankravets, 2016-03-22:
Could you reproduce this issue again with `pio lib install 1`?

nogurenn, 2016-03-22:
Here's a screenshot just now for your request.
[image]

ivankravets, 2016-03-22:
Try in that Terminal
```
curl "http://dl.platformio.org/libraries/archives/0/2953.tar.gz"
```

Does it show any errors?

nogurenn, 2016-03-22:
Well, no errors, but I don't know what to make of this dump. Also, `pip install -U platformio` work if I execute in on my computer's native Terminal. Unfortunately, `pio lib install 1` doesn't work on my native terminal, generating the same error.
[image]

ivankravets, 2016-03-22:
Thanks. Click on `Settings(gear icon) > Packages > Community Packages > platformio-ide` and click `Use development version`. 

Try `pio lib install 1`.

ivankravets, 2016-03-22:
I forget, after click on `Use development version`, please wait for a while (it will install develop version in the background).

ivankravets, 2016-03-22:
Also, please check via `pio --version` that 2.8.6.dev is installed.

nogurenn, 2016-03-22:
After enabling `Use development version`, `pio --version` is still `2.8.5`. The error persists

ivankravets, 2016-03-22:
You have 2 version `platformio` now: 1 is installed in IDE and second is installed globally in the system

You write that problem persists in native Terminal too. Please open native Terminal and install the latest development version http://docs.platformio.org/en/latest/installation.html#development-version. Please check version before lib installing.

P.S: It's important for us because we are going to release PlatformIO 2.8.6 and new IDE through 15-20mins. Thanks a lot for the help.

nogurenn, 2016-03-22:
Am I doing something wrong?

About the native terminal `platformio`, I double-checked it, and it appears that they're using the same  `pio`. My apologies. Is this supposed to be normal though?

[image]

ivankravets, 2016-03-22:
Could you provide me TeamViewer session? Contact me@ikravets.com with Session ID/Pass. Thanks!

nogurenn, 2016-03-22:
@ivankravets As much as I'd like to help right away, I don't know how to use TeamViewer... What's the cause of my issue?

ivankravets, 2016-03-22:
I need temporary access to your Mac to debug this issue under your control. Please install https://www.teamviewer.com and send me@ikravets.com Session ID/Pass. I'll have temporary access to your Mac while you close the TeamViewer application.

commit d83b1280 (2016-03-22): Fix "KeyError: 'content-length'" in PlatformIO Download Manager // Resolve #591
closed 2016-03-22 by ivankravets