platformio/platformio-core #592: ESP8266 fast flashing

georgkreimer, 2016-03-22:
What kind of issue is this?

- [ ] Feature Request: Faster flashing for ESP8266 (and CC3200) boards

------------------------------------------------------------------

### Configuration

**Operating system**: Any supported

**PlatformIO Version** (`platformio --version`): Any

### Description of improvement

Flashing of the ESP8266 boards takes a while. Anybody who has flashed some of them noticed it's about a minute. I saw [this post](https://blog.cesanta.com/esp8266-superfast-flashing-flash-esp-in-5-seconds) blog today. Those guys at Cesanta Software have created a [modified version of the esptool](https://github.com/cesanta/fnc/blob/master/common/platforms/esp8266/stubs/esptool.py) but also have another flash tool called [Flash'N'Chips](https://github.com/cesanta/fnc) which reduces the flash times significantly. It would be great to have the flash time reduced for PlatformIO using these 2 options.

ivankravets, 2016-03-22:
> Anybody who has flashed some of them noticed it's about a minute.
What is your board? Do you use specific LD script?

georgkreimer, 2016-03-25:
Just verified with huzzah and NodeMCU breakout boards. It's around 40 sec for me. I'm using the default settings with the PlatformIO IDE

Jorgen-VikingGod, 2016-03-25:
maybe this article could help to improve the flashing speed
https://blog.cesanta.com/esp8266-superfast-flashing-flash-esp-in-5-seconds

eni23, 2016-08-21:
+1 for the idea that @Jorgen-VikingGod suggested. Using the modified esptool.py from the Mongoose platform (Source: https://github.com/cesanta/mongoose-flashing-tool/blob/master/common/platforms/esp8266/stubs/esptool.py) speeds up flashing a lot compared to the one shipped with PlatformIO:

PlatformIO esptool: `0m16.152s`
Modded esptool.py: `0m4.345s`

Until this or another fast flasher is integrated, one could do a workaround like this (only tested with PlatformIO 3):

Place the modded esptool to `tools/esptool-cesanta.py`

`platformio.ini`
```ini
[env:nodemcuv2]
platform = espressif
framework = arduino
board = nodemcuv2
upload_speed = 3000000
extra_script = tools/custom_uploader.py
```

`tools/custom_uploader.py`
```python
import os
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()
esptool_mod = "{0}/tools/esptool-cesanta.py".format(env['PROJECT_DIR'])
env.Replace(UPLOADER=esptool_mod)
env.Replace(UPLOADERFLAGS=[
    '--port',
    '$UPLOAD_PORT',
    'write_flash',
    '--flash_baud=$UPLOAD_SPEED',
    '0'
])
env.Replace(UPLOADCMD='$UPLOADER $UPLOADERFLAGS $SOURCE')
```

ivankravets, 2016-11-27:
Moved to https://github.com/platformio/platform-espressif8266/issues/15

closed 2016-11-27 by ivankravets