platformio/platformio-core #594: Flags specified in platformio.ini build_flags variable not applied to all necessary steps in the build process

apvanzanten, 2016-03-23:
### Configuration

**PlatformIO version 2.8.6**

**Platform: ststm32 - disco_f407vg**

### Description of problem
When supplying platformio with flags via the build_flags variable in the platformio.ini file, these flags are applied only to the compilation of the user-supplied source (src and lib folder) but not in the final linkage. 

#### Steps to Reproduce
I have set up a test case [here](https://github.com/apvanzanten/platformioCompilerFlagsBugTestcase) on github, the readme on it explains how to use it (it is very simple, basically just clone and platformio run). 

### Actual Results
The test case tries to build using flags for hardware floating point, which are applied as expected on the compilation of the main.c file, but then not anywhere else in the build process, this causes problems when the linker tries to link the hard-float compiled main.o to the soft-float compiled firmware.elf.

### Expected Results
For the build flags to be applied properly and the project to build without errors.

valeros, 2016-03-23:
Hi @apvanzanten! Which framework are you going to use?

apvanzanten, 2016-03-23:
I am currently using spl. However for the test case I configured it for mbed because that was the default and because it gave a cleaner and simpler test case. 

I slightly modified the test case so it works for spl and put it in a [separate branch of my repo](https://github.com/apvanzanten/platformioCompilerFlagsBugTestcase/tree/splCase).

valeros, 2016-03-23:
I can suggest such workaround:
platformio.ini:
```ini
[env:disco_f407vg]
platform = ststm32
framework = spl
board = disco_f407vg
build_flags = -mthumb -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16
extra_script = update_link_flags.py
```
update_link_flags.py:
```python
# Custom settings, as referred to as "extra_script" in platformio.ini
#
# See http://docs.platformio.org/en/latest/projectconf.html#extra-script

from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Append(
    LINKFLAGS=[
        "-mthumb",
        "-march=armv7e-m",
        "-mfloat-abi=hard",
        "-mfpu=fpv4-sp-d16"
    ]
)
```

apvanzanten, 2016-03-23:
That works a treat. Thanks!

valeros, 2016-03-23:
Thanks a lot for the report! Happy Coding!

Please give a star :star: for our repositories:
- https://github.com/platformio/platformio-atom-ide/stargazers
- https://github.com/platformio/platformio/stargazers

Thanks!

closed 2016-03-23 by valeros

mtrocki, 2017-02-21:
Hello, I have the same problem but I'm using Platform: stm32f0discovery.
Do I have to create update_link_flags.py?

positron96, 2021-12-24:
Hi. Why is this closed? I'm having the same problem with stm32f407 and it does not seem that this is fixed.

ivankravets, 2021-12-24:
@positron96 what is your platformio.ini?

positron96, 2021-12-27:
Something in the lines of this:
```
[platformio]

src_dir = .

[env]
platform = ststm32@14.2.0

extra_scripts = extra-build.py

build_flags = 
    -mfpu=fpv4-sp-d16
    -mfloat-abi=hard

src_filter =     
; a lot of sources

[env:stm]
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTx_FLASH.ld
upload_protocol = jlink
debug_tool = jlink

```

`mfpu` and `mfloat-eabi` don't get passed to linker, so I had to manually patch that with a script

ivankravets, 2021-12-27:
> so I had to manually patch that with a script

Sorry, this is a valid solution at this time. We will work on https://github.com/platformio/platformio-core/issues/2144