platformio/platformio-core #595: Python 3.x support

DavidPowell, 2016-03-24:
PlatformIO seems like a really innovative new tool. Thus it is rather surprising that it forces users to install Python 2.7, which will be unsupported in a few years. So I have a couple of questions (which I intend to be constructive, not as a criticism of the developers):

- Is any effort currently under way to make it compatible with Python 3.x?

- Are there any known obstacles (e.g. Python 2.x only libraries) to moving to a single code-base supporting both Python 2.x/3.x?

- Would you welcome any efforts to address Py3 compatibility?

Obijuan, 2016-03-24:
It think one of the blockers is scons, that is only available for python 2.7. But in [this link](http://scons.org/scons-241-is-available.html) they say: "This will be the last release to support Python versions earlier than 2.7, as we begin to move toward supporting Python 3."

I also think that it would be great if platformio is ported to python3

ivankravets, 2016-03-24:
As @Obijuan has answered above, PlatformIO depends on [SCons](http://scons.org). When SCons will support Python 3, we also update our code.

closed 2016-03-24 by ivankravets

cfobel, 2018-02-13:
See also #895