platformio/platformio-core #596: Can't upload code to seeeduino Arch Max

sstaub, 2016-03-25:
I can't upload code to my Seeeduino Arch Max. Here a screenshot with the error code. OS X 10.10.5
Last version of Plarformio, IDE and mbed framework. I can compile the code on the mbed online IDE.
[image]

ivankravets, 2016-03-25:
Mac OS doesn't allow via GCC to build program with `-static` libs. Temporary solution:
1. Install [Homebrew](http://brew.sh)
2. `$ brew install libusb`

sstaub, 2016-03-26:
I installed libusb via macports, but it doesn't work.

sstaub, 2016-03-26:
Maybe that is problem with OS X yosemite.

ivankravets, 2016-03-26:
Please try `brew`

sstaub, 2016-03-26:
I installed with homebrew, but a get another error. The board has the last firmware installed.
[image]

ivankravets, 2016-03-27:
I've just rebuilt `tool-stlink` for Mac. Please download [tool-stlink](http://cl.ly/1o071b2E3k3r) manually. Unzip it. Replace with files in `/Users/sstaub/.platformio/packages/tool-stlink`. 

If it works for you, I'll update package globally.

sstaub, 2016-03-27:
It doesn't work, I get the same error again. Maybe that is a board specific problem. I can upload
to a STM32F303 Nucleo board without a problem.
I think on installation for mac, the 'libusb' should also installed.

ivankravets, 2016-03-27:
@valeros do you have any ideas?

sstaub, 2016-03-27:
I think the upload must work like on other mbed boards like the Arch Pro. The code is uploaded to USB-drive 'mbed', It seems that the ST-Link don't use this way.

ivankravets, 2016-03-27:
Do you have `mbed` disk drive?

sstaub, 2016-03-27:
Yes. The board doesn't have an ST-Link debugger. See: http://www.seeedstudio.com/wiki/Arch_Max

ivankravets, 2016-03-27:
It looks strange because mbed should be used. See https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/ststm32.py#L106

Please contact me@ikravets.com with TeamViewer session.

ivankravets, 2016-03-29:
Fixed in https://github.com/platformio/platformio/commit/bc3d8d26d0f4cbed51355aa0111988de87afc822

closed 2016-03-29 by ivankravets

thinkyhead, 2021-08-04:
Strangely, now seeing this with tool-dfuutil 1.9.200310.

ivankravets, 2021-08-05:
Could you try `brew install libusb`?

alexgeron-Luos, 2021-09-21:
Hello @ivankravets i have the same issue (on M1 macbook) but the 

`brew install libusb` didn't work for me.

When i try to compile my App i get the following issue 

`dyld: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: /Users/alexisgeron/.platformio/packages/tool-dfuutil/bin/dfu-suffix
  Reason: no suitable image found.  Did find:
        /usr/local/opt/libusb/lib/libusb-1.0.0.dylib: mach-o, but wrong architecture
        /usr/local/opt/libusb/lib/libusb-1.0.0.dylib: stat() failed with errno=1
        /opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib: mach-o, but wrong architecture
*** [.pio/build/l0/firmware.bin] Error -6`