platformio/platformio-core #598: SAMD compilation appears broken

malachib, 2016-03-27:
I've tried compiling many different test programs including a simple blinky, but all give a whole slew of errors not finding tons of symbols, starting with:

"In file included from .pioenvs/spark_samd/FrameworkArduino/Arduino.h:61:0,
from src/main.cpp:1:
.pioenvs/spark_samd/FrameworkArduino/WVariant.h:93:36: error: 'TCC_INST_NUM' was not declared in this scope"

and including other symbols such as Sercom, UsbHostDescriptor, and a lot more

Testing done on Debian Jessie and Mac OS X

valeros, 2016-03-28:
Thanks! Fixed in the new framework package.  Please update via `pio update`.

ivankravets, 2016-03-28:
Please test development version http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-03-28 by ivankravets

malachib, 2016-03-28:
Compiling once again! Thank you guys