platformio/platformio-core #603: platfofmio installs, but does not work under linux(Ubuntu 16.04 x64)

Pilleo, 2016-04-04:
- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 16.04 x64

**PlatformIO Version** (`platformio --version`): latest installed from pip or your script

### Description of problem
Here is output of your script and --help:
`leonid@leonid-Lenovo-G500:~$ sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
[sudo] password for leonid: 

==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /usr/lib/python2.7/dist-packages

[SUCCESS]

==> Installing PlatformIO and dependencies ...
Requirement already up-to-date: platformio in ./.local/lib/python2.7/site-packages
Requirement already up-to-date: click<6,>=3.2 in ./.local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: requests<3,>=2.4.0 in ./.local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial<4 in ./.local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: colorama in ./.local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: bottle<0.13 in ./.local/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: lockfile<0.13,>=0.9.1 in ./.local/lib/python2.7/site-packages (from platformio)

[SUCCESS]

 ==> Installation process has been successfully FINISHED! <==

Usage: __main__.py [OPTIONS] COMMAND [ARGS]...

Options:
  --version          Show the version and exit.
  -f, --force        Force to accept any confirmation prompts.
  -c, --caller TEXT  Caller ID (service).
  -h, --help         Show this message and exit.

Commands:
  boards       Pre-configured Embedded Boards
  ci           Continuous Integration
  init         Initialize new PlatformIO based project
  lib          Library Manager
  platforms    Platforms and Packages Manager
  run          Process project environments
  serialports  List or Monitor Serial ports
  settings     Manage PlatformIO settings
  update       Update installed Platforms, Packages and Libraries
  upgrade      Upgrade PlatformIO to the latest version

----------------------------------------
Please RESTART your Terminal Application
----------------------------------------

Then run `platformio --help` command.

leonid@leonid-Lenovo-G500:~$ platformio --help
platformio: command not found
`
Devoit is not working also, but it says it installs successfully. 
#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2016-04-04:
I see that `pip` installs packages in the ` ./.local/`. Does ` ./.local/` contain `bin` folder? If yes, is this folder added to `$PATH`?

Pilleo, 2016-04-04:
./.local/ does contain bin folder. But i do not understand how to check or modify $PATH .

ivankravets, 2016-04-04:
@Pilleo Looks like Ubuntu 16 has significantly changes. 

Don't forget to add to your bash profile because after restart you will need again to set $PATH

ivankravets, 2016-04-04:
See http://unix.stackexchange.com/questions/26047/how-to-correctly-add-a-path-to-path

Pilleo, 2016-04-04:
Well, now it does not work at all. I was trying to modify .profile, and did not work. I restored it, and tried 
again ` PATH=$PATH:/.local/bin export PATH`, but this time it does not work also...

orgkhnargh, 2016-04-04:
Are you sure about `PATH=$PATH:/.local/bin` part? Shouldn't there be another dot between `:` and `/.`?

orgkhnargh, 2016-04-04:
@Pilleo, be aware that `./.local/bin` is a relative path and it will only work while you are in home directory. I advise you to change the line to this: `export PATH=$PATH:$HOME/.local/bin`.

Pilleo, 2016-04-04:
It think it is ok now, finally!) 
added `export PATH=$PATH:$HOME/.local/bin` to .profile
Thank you very much!

closed 2016-04-04 by Pilleo