platformio/platformio-core #606: IOError: [Errno 13] Permission denied

reauv, 2016-04-06:
Hey there,

I've been trying to run a build for a NodeMCU which requires the espressif platform.
But after that I get an error `IOError: [Errno 13] Permission denied`

### Configuration

**Operating system**: OS X 10.11.4

**PlatformIO Version**: 2.8.6

### Description of problem
Trying to run a build for a NodeMCU, but after attempting to install some dependencies I get a permission denied error.

#### Steps to Reproduce

1. `platformio init --board nodemcuv2`
2. `platformio run`
3. Press y when asked to `espressif` 

### Actual Results
```
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/run.py", line 90, in cli
    results.append(ep.process())
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/run.py", line 126, in process
    result = self._run()
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/run.py", line 192, in _run
    return p.run(build_vars, build_targets, self.verbose_level)
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 426, in run
    self._install_default_packages()
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 412, in _install_default_packages
    return self.install()
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 329, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 113, in install
    if self.unpack(dlpath, pkg_dir):
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 52, in unpack
    return fu.start()
  File "/usr/local/lib/python2.7/site-packages/platformio/unpacker.py", line 103, in start
    self._unpacker.extract_item(item, self._dest_dir)
  File "/usr/local/lib/python2.7/site-packages/platformio/unpacker.py", line 36, in extract_item
    self._afo.extract(item, dest_dir)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tarfile.py", line 2115, in extract
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name))
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tarfile.py", line 2191, in _extract_member
    self.makefile(tarinfo, targetpath)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tarfile.py", line 2231, in makefile
    with bltn_open(targetpath, "wb") as target:
IOError: [Errno 13] Permission denied: '/Users/Robert/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/cassert'
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[env:nodemcuv2]
platform = espressif
framework = arduino
board = nodemcuv2
```

### Additional info
I've installed PlatformIO with `sudo pip --no-cache-dir install -U platformio` as described [here](https://github.com/platformio/platformio/issues/295#issuecomment-143772005)

ivankravets, 2016-04-06:
1. Please remove totally `~/.platformio` directory
2. DON'T run `platformio` command with `sudo`.

reauv, 2016-04-06:
Ah ok, I guess tried running sudo after getting errors. I've removed the `~/.platformio` folder as you said and after running `platformio run` again I'm getting the following error stack.

```
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/run.py", line 90, in cli
    results.append(ep.process())
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/run.py", line 126, in process
    result = self._run()
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/run.py", line 192, in _run
    return p.run(build_vars, build_targets, self.verbose_level)
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 426, in run
    self._install_default_packages()
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 412, in _install_default_packages
    return self.install()
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 329, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 113, in install
    if self.unpack(dlpath, pkg_dir):
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 52, in unpack
    return fu.start()
  File "/usr/local/lib/python2.7/site-packages/platformio/unpacker.py", line 103, in start
    self._unpacker.extract_item(item, self._dest_dir)
  File "/usr/local/lib/python2.7/site-packages/platformio/unpacker.py", line 36, in extract_item
    self._afo.extract(item, dest_dir)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tarfile.py", line 2115, in extract
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name))
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tarfile.py", line 2199, in _extract_member
    self.makelink(tarinfo, targetpath)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tarfile.py", line 2276, in makelink
    os.unlink(targetpath)
OSError: [Errno 1] Operation not permitted: '/Users/Robert/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/lib64'
```

So different error code this time: `OSError: [Errno 1] Operation not permitted`

ivankravets, 2016-04-06:
1. Give me please an output of `ls -l ~/.platformio`
2. Do you have correct permissions to `~/.platformio` folder?

mbernson, 2016-04-06:
I'm having the same issue, also on Mac OS X 10.11.4 with the nodemcuv2 board. Installing toolchains for other boards works fine though.

I tried downloading the xtensa toolchain myself (from the PlatformIO manifest URL) and extracting it in the Finder results in the same "Error 1 - Operation not permitted". Extracting it via the command line doesn't work either:

```
$ wget https://dl.bintray.com/platformio/dl-packages/toolchain-xtensa-darwin_x86_64-1.40802.0.tar.gz
$ tar -xzf toolchain-xtensa-darwin_x86_64-1.40802.0.tar.gz
bin/xtensa-lx106-elf-readelf: Cannot extract through symlink bin
bin/xtensa-lx106-elf-ld.bfd: Cannot extract through symlink bin
bin/xtensa-lx106-elf-gcc-ar: Cannot extract through symlink bin
bin/xtensa-lx106-elf-strings: Cannot extract through symlink bin
bin/xtensa-lx106-elf-gcc-4.8.2: Cannot extract through symlink bin
bin/xtensa-lx106-elf-objcopy: Cannot extract through symlink bin
bin/xtensa-lx106-elf-gcc-ranlib: Cannot extract through symlink bin
bin/xtensa-lx106-elf-elfedit: Cannot extract through symlink bin
bin/xtensa-lx106-elf-cpp: Cannot extract through symlink bin
bin/xtensa-lx106-elf-g++: Cannot extract through symlink bin
bin/xtensa-lx106-elf-gdb: Cannot extract through symlink bin
bin/xtensa-lx106-elf-gcc: Cannot extract through symlink bin
bin/xtensa-lx106-elf-gcov: Cannot extract through symlink bin
bin/xtensa-lx106-elf-c++filt: Cannot extract through symlink bin
bin/xtensa-lx106-elf-objdump: Cannot extract through symlink bin
bin/xtensa-lx106-elf-c++: Cannot extract through symlink bin
bin/xtensa-lx106-elf-size: Cannot extract through symlink bin
bin/xtensa-lx106-elf-nm: Cannot extract through symlink bin
xtensa-lx106-elf/lib64: Can't remove already-existing dir
bin/xtensa-lx106-elf-ld: Cannot extract through symlink bin
tar: Error exit delayed from previous errors.
```

I guess the xtensa archive file is broken?

ivankravets, 2016-04-06:
This is our problem. We are updating packages with the new structures for PlatformIO 3.0. The previous PaltformIO 2.0 packages have been restored.

Please try again `rm -r ~/.platformio`. It should work.

Sorry.

mbernson, 2016-04-06:
It works again, thanks @ivankravets!

ivankravets, 2016-04-06:
@mbernson thanks for the confirmation.

@reauv does it work for you?

reauv, 2016-04-06:
Yes, works like a charm. Thanks for the quick help @ivankravets!

closed 2016-04-06 by reauv

kumarkishansingh, 2017-06-05:
IOError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/scandir.pyc'
There is no scandir.pyc in  '/usr/local/lib/python2.7/dist-packages' ....How to fix it ? Please  answer as soon as possible...