platformio/platformio-core #607: teensy31 build failure on Raspberry Pi 2 - "cannot find -larm_cortexM4l_math"

blackketter, 2016-04-06:
### Configuration

**Operating system**:
Raspian Jessie 

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.8.6

### Description of problem
Build of teensy31 blink example fails with:

```
arm-none-eabi-g++ -o .pioenvs/teensy31/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 -Wl,--defsym=__rtc_localtime=1459954039 -fsingle-precision-constant --specs=nano.specs -Wl,-T"mk20dx256.ld" .pioenvs/teensy31/src/tmp_ino_to.o -L/home/pi/.platformio/packages/ldscripts -L.pioenvs/teensy31 -Wl,--start-group -larm_cortexM4l_math .pioenvs/teensy31/libFrameworkArduino.a -lc -lgcc -lm -Wl,--end-group
/home/pi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: cannot find -larm_cortexM4l_math
collect2: error: ld returned 1 exit status
```

Blink.ino:

```
/* LED Blink, Teensyduino Tutorial #1
   http://www.pjrc.com/teensy/tutorial.html
 
   This example code is in the public domain.
*/

// Teensy 2.0 has the LED on pin 11
// Teensy++ 2.0 has the LED on pin 6
// Teensy 3.0 has the LED on pin 13
const int ledPin = 11;

// the setup() method runs once, when the sketch starts

void setup() {
  // initialize the digital pin as an output.
  pinMode(ledPin, OUTPUT);
}

// the loop() methor runs over and over again,
// as long as the board has power

void loop() {
  digitalWrite(ledPin, HIGH);   // set the LED on
  delay(1000);                  // wait for a second
  digitalWrite(ledPin, LOW);    // set the LED off
  delay(1000);                  // wait for a second
}

```

Platformio.ini:
```
[env:teensy31]
platform = teensy
framework = arduino
board = teensy31
```

valeros, 2016-04-06:
Hi @blackketter ! Thanks for the report!
Could you please add this file [libarm_cortexM4l_math.a](https://dl.dropboxusercontent.com/u/10025728/libarm_cortexM4l_math.zip) to your project and specify the path to this library in `platformio.ini` :
```ini
[env:teensy31]
platform = teensy
framework = arduino
board = teensy31
build_flags = -L/path/to/libarm_cortexM4l_math
```
Does it work?

blackketter, 2016-04-06:
That did it, thanks!

valeros, 2016-04-07:
Fixed in toolchain package.

closed 2016-04-07 by valeros