platformio/platformio-core #608: Raspberry Pi 1 Model B build for Teensy31 fails with missing nano.specs

blackketter, 2016-04-06:
### Configuration

**Operating system**:
Raspian Jesse, RasPi 1 Model B

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.8.6

### Description of problem
Fresh install of platformio on a Raspberry Pi 1 Model B fails to build blink example with this error:

```
[Wed Apr  6 12:33:02 2016] Processing teensy31 (platform: teensy, board: teensy31, framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/teensy31/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 -Wl,--defsym=__rtc_localtime=1459974787 -fsingle-precision-constant --specs=nano.specs -Wl,-T"mk20dx256.ld" .pioenvs/teensy31/src/tmp_ino_to.o -L/home/pi/.platformio/packages/ldscripts -L.pioenvs/teensy31 -Wl,--start-group -larm_cortexM4l_math .pioenvs/teensy31/libFrameworkArduino.a -lc -lgcc -lm -Wl,--end-group
arm-none-eabi-g++: error: nano.specs: No such file or directory
scons: *** [.pioenvs/teensy31/firmware.elf] Error 1
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:teensy31]
platform = teensy
framework = arduino
board = teensy31
```

**Source file to reproduce issue:**
```cpp
/* LED Blink, Teensyduino Tutorial #1
   http://www.pjrc.com/teensy/tutorial.html

   This example code is in the public domain.
*/

// Teensy 2.0 has the LED on pin 11
// Teensy++ 2.0 has the LED on pin 6
// Teensy 3.0 has the LED on pin 13
const int ledPin = 13;

// the setup() method runs once, when the sketch starts

void setup() {
  // initialize the digital pin as an output.
  pinMode(ledPin, OUTPUT);
}

// the loop() methor runs over and over again,
// as long as the board has power

void loop() {
  digitalWrite(ledPin, HIGH);   // set the LED on
  delay(100);                  // wait for a second
  digitalWrite(ledPin, LOW);    // set the LED off
  delay(100);                  // wait for a second
}
```

### Additional info
I did not see this error on a Raspberry Pi 2.  The nano.specs files do appear there, but not on the failing system:

```
grep -ir ~/.platformio nano.specs
grep: nano.specs: No such file or directory
ls ~/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/lib/
crt0.o         libarm_cortexM0l_math.a   libc.a            libm.a       librdpmon.a         libstdc++.la  libsysbase.a  pid.specs      rdpmon-crt0.o   redboot.ld          thumb
iq80310.specs  libarm_cortexM4lf_math.a  libg.a            libnosys.a   libstdc++.a         libsupc++.a   linux-crt0.o  rdimon-crt0.o  rdpmon.specs    redboot.specs
ldscripts      libarm_cortexM4l_math.a   libgloss-linux.a  librdimon.a  libstdc++.a-gdb.py  libsupc++.la  linux.specs   rdimon.specs   redboot-crt0.o  redboot-syscalls.o
```

valeros, 2016-04-07:
Thanks, fixed in the new toolchain package, please update via `platformio update`

blackketter, 2016-04-07:
Works, great, thanks!

blackketter, 2016-04-07:
Ok, so it builds now, but the resulting binary does not run.  Building the same project (same platformio.ini and blink.cpp file) on Mac and Raspberry Pi 2 result in working firmware.hex of the same size (10676 bytes), but when building on Raspberry Pi 1, the firmware.hex is only 11080 bytes.  Let me know if you need any of the files.

blackketter, 2016-04-07:
I've compared the build commands as sent to the console and they are identical with the following weird exception.  The warnings below are only generated on the RPi 1, but not on the RPi2 or Mac:
```
arm-none-eabi-gcc -o .pioenvs/teensy31/FrameworkArduino/mk20dx128.o -c -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/teensy31/FrameworkArduino .pioenvs/teensy31/FrameworkArduino/mk20dx128.c
.pioenvs/teensy31/FrameworkArduino/mk20dx128.c: In function 'ResetHandler':
.pioenvs/teensy31/FrameworkArduino/mk20dx128.c:588:18: warning: initialization from incompatible pointer type [enabled by default]
uint32_t *src = &_etext;
^
.pioenvs/teensy31/FrameworkArduino/mk20dx128.c:589:19: warning: initialization from incompatible pointer type [enabled by default]
uint32_t *dest = &_sdata;
^
.pioenvs/teensy31/FrameworkArduino/mk20dx128.c:653:14: warning: comparison of distinct pointer types lacks a cast [enabled by default]
while (dest < &_edata) *dest++ = *src++;
^
.pioenvs/teensy31/FrameworkArduino/mk20dx128.c:654:7: warning: assignment from incompatible pointer type [enabled by default]
dest = &_sbss;
^
.pioenvs/teensy31/FrameworkArduino/mk20dx128.c:655:14: warning: comparison of distinct pointer types lacks a cast [enabled by default]
while (dest < &_ebss) *dest++ = 0;
^
```

valeros, 2016-04-08:
@blackketter Please retest with the latest toolchain package.

blackketter, 2016-04-08:
That seemed to fix it!  Thanks.

closed 2016-04-08 by blackketter