platformio/platformio-core #609: Teensy 3.2 cannot be programmed after first time, unless press reset button

424778940z, 2016-04-07:
I'm using latest stable version platformio ide

This issue may caused by I set it to usb uart mode, and download script doesn't reboot it before program.

i did following change in this file: .atom\packages\platformio-ide\penv\Lib\site-packages\platformio\builder\scripts\teensy.py
start at: Line 68

before:
`    env.Append(
        UPLOADER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_post_compile"),
        UPLOADERFLAGS=[
            "-file=firmware",
            '-path="$BUILD_DIR"',
            '-tools="%s"' % join("$PIOPACKAGES_DIR", "tool-teensy")
        ],
        UPLOADHEXCMD='"$UPLOADER" $UPLOADERFLAGS'
    )`

after:
`    env.Append(
		REBOTTER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_reboot"),
        UPLOADER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_post_compile"),
        UPLOADERFLAGS=[
            "-file=firmware",
            '-path="$BUILD_DIR"',
            '-tools="%s"' % join("$PIOPACKAGES_DIR", "tool-teensy")
        ],
        UPLOADHEXCMD='"$REBOTTER" & "$UPLOADER" $UPLOADERFLAGS'
    )`

it solve this issue for now but will have a error because delay between restart and program is not enough
error msg: "Teensy Loader is currently busy with another operation (p).  Please try again in a few seconds, or restart Teensy Loader."

ivankravets, 2016-04-07:
@PaulStoffregen Could you explain upload scenario? Should we make "teensy_reboot" before uploading? Currently, Teensy CLI asks to press "Reset button...".

PaulStoffregen, 2016-04-07:
In Arduino, we run teensy_post_compile first, which sends a message to the (GUI-based) Teensy Loader to open the latest compiled file, and configure in auto mode.  Then teensy_reboot is run, to send a request to the Teensy to reboot into bootloader mode.  Because Teensy Loader is in auto mode, when the Teensy appears, it gets programmed by Teensy Loader.

If you're using the CLI instead of GUI, you probably want to run teensy_reboot first.

424778940z, 2016-04-07:
@PaulStoffregen thanks for the information, current official script using GUI loader, but not reboot

424778940z, 2016-04-07:
`		REBOTTER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_reboot"),
        UPLOADER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_post_compile"),
        UPLOADERFLAGS=[
            "-file=firmware",
            '-path="$BUILD_DIR"',
            '-tools="%s"' % join("$PIOPACKAGES_DIR", "tool-teensy")
        ],
        UPLOADHEXCMD='"$UPLOADER" $UPLOADERFLAGS  & "$REBOTTER"'`

This is the final code, works perfectly.

ivankravets, 2016-04-09:
@424778940z by default PlatformIO uses Teensy Loader CLI. If you want to use Teensy Loader GUI, please place all files to `~/.platformio/packages/tool-teensy`:
```
/Users/ikravets/.platformio/packages/tool-teensy
├── teensy.app
├── teensy_gateway
├── teensy_post_compile
├── teensy_reboot
└── teensy_restart
```

` teensy_post_compile` will call `teensy_reboot` automatically. We run the same comand as @PaulStoffregen does in Arduino IDE. See

Arduino IDE
```
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/..//teensy_post_compile" "-file=Blink.ino" "-path=/var/folders/nj/nt38kxwd2xxfl5l42np_w7qw0000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp" "-tools=/Applications/Arduino.app/Contents/Java/hardware/tools/avr/../" "-board=TEENSY31"
Opening Teensy Loader...
```

PlatformIO
```
"/Users/ikravets/.platformio/packages/tool-teensy/teensy_post_compile" -file=firmware -path="/Volumes/SOFT/Projects/GitHub/platformio/origin/examples/wiring-blink/.pioenvs/teensy31" -tools="/Users/ikravets/.platformio/packages/tool-teensy"
Opening Teensy Loader...
```

424778940z, 2016-04-09:
@ivankravets 

em...that's strange. I install platformio IDE with touch any settings. it use GUI version by default ...
And teensy_post_compile never call teensy_reboot....

ivankravets, 2016-04-09:
@424778940z  Do you see `teensy_loader_cli` in  `~/.platformio/packages/tool-teensy`?

424778940z, 2016-04-09:
yes, it is there

ivankravets, 2016-04-09:
@424778940z Sorry, I've just remembered it... Teensy Loader CLI doesn't work on Windows :disappointed: See http://www.pjrc.com/teensy/loader_cli.html

You can ask @PaulStoffregen to add support in this repo https://github.com/PaulStoffregen/teensy_loader_cli

ivankravets, 2016-04-09:
@PaulStoffregen we call `teensy_post_compile` for uploading... Should we call separately `teensy_reboot`? I don't see this call in Teensyduino IDE.

424778940z, 2016-04-09:
@ivankravets PaulStoffregen is already answered above.

This issue is already solved by modify platformio script.

However, my way is a little bit lazy, just use & run them at one line.
Put them separately may looks better.

ivankravets, 2016-04-09:
@424778940z I don't understand how does it work for you. Rebooter requires GUI to be run. In your case, you call Rebooter before Teensy GUI.

Ok, I'll implement it.

424778940z, 2016-04-09:
@ivankravets 
At the first time GUI start will looking for the ready board, then upload it and reset. This is works perfectly in platformio.

Problem is about second time upload, because GUI is already running, according to PaulStoffregen, it supposed to call teensy_post_compile fist tell GUI new file is ready, then call teensy_reboot reboot the hardware to bootloader mode allows GUI upload again.

In current platformio official code, it only call teensy_post_compile, but no reboot. Therefore GUI knows there is a new file ready, but the hardware is not in bootloader mode, GUI cannot upload, if people press reset button on the board at this time, it will uoload.

424778940z, 2016-04-09:
Here is the whole modified script

```python
# Copyright 2014-2016 Ivan Kravets <me@ikravets.com>
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#    http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""
    Builder for Teensy boards
"""

from os.path import isfile, join

from SCons.Script import (COMMAND_LINE_TARGETS, AlwaysBuild, Default,
                          DefaultEnvironment, SConscript)

env = DefaultEnvironment()

if env.get("BOARD_OPTIONS", {}).get("build", {}).get("core") == "teensy":
    SConscript(env.subst(join("$PIOBUILDER_DIR", "scripts", "baseavr.py")))
elif env.get("BOARD_OPTIONS", {}).get("build", {}).get("core") == "teensy3":
    SConscript(env.subst(join("$PIOBUILDER_DIR", "scripts", "basearm.py")))
    env.Append(
        LINKFLAGS=[
            "-Wl,--defsym=__rtc_localtime=$UNIX_TIME",
            "-fsingle-precision-constant",
            "--specs=nano.specs"
        ],
        CCFLAGS=[
            "-fsingle-precision-constant"
        ]
    )

env.Append(
    CPPDEFINES=[
        "USB_SERIAL",
        "LAYOUT_US_ENGLISH"
    ],

    CXXFLAGS=[
        "-std=gnu++0x",
        "-felide-constructors"
    ]
)

if isfile(env.subst(join(
        "$PIOPACKAGES_DIR", "tool-teensy", "teensy_loader_cli"))):
    env.Append(
        UPLOADER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_loader_cli"),
        UPLOADERFLAGS=[
            "-mmcu=$BOARD_MCU",
            "-w",  # wait for device to apear
            "-s",  # soft reboot if device not online
            "-v"   # verbose output
        ],
        UPLOADHEXCMD='"$UPLOADER" $UPLOADERFLAGS $SOURCES'
    )
else:
    env.Append(
	#BZH_FIXED
        # UPLOADER=join(
            # "$PIOPACKAGES_DIR", "tool-teensy", "teensy_post_compile"),
        # UPLOADERFLAGS=[
            # "-file=firmware",
            # '-path="$BUILD_DIR"',
            # '-tools="%s"' % join("$PIOPACKAGES_DIR", "tool-teensy")
        # ],
        # UPLOADHEXCMD='"$UPLOADER" $UPLOADERFLAGS'
		REBOTTER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_reboot"),
        UPLOADER=join(
            "$PIOPACKAGES_DIR", "tool-teensy", "teensy_post_compile"),
        UPLOADERFLAGS=[
            "-file=firmware",
            '-path="$BUILD_DIR"',
            '-tools="%s"' % join("$PIOPACKAGES_DIR", "tool-teensy")
        ],
        UPLOADHEXCMD='"$UPLOADER" $UPLOADERFLAGS  & "$REBOTTER"'
    )

#
# Target: Build executable and linkable firmware
#

target_elf = env.BuildProgram()

#
# Target: Build the firmware file
#

if "uploadlazy" in COMMAND_LINE_TARGETS:
    target_firm = join("$BUILD_DIR", "firmware.hex")
else:
    target_firm = env.ElfToHex(join("$BUILD_DIR", "firmware"), target_elf)

#
# Target: Print binary size
#

target_size = env.Alias("size", target_elf, "$SIZEPRINTCMD")
AlwaysBuild(target_size)

#
# Target: Upload by default firmware file
#

upload = env.Alias(["upload", "uploadlazy"], target_firm, "$UPLOADHEXCMD")
AlwaysBuild(upload)

#
# Target: Define targets
#

Default([target_firm, target_size])

```

ivankravets, 2016-04-09:
@424778940z Please re-test 2.9.0.dev0 http://docs.platformio.org/en/latest/installation.html#development-version

424778940z, 2016-04-09:
@ivankravets Tested, it works perfectly.

ivankravets, 2016-04-09:
@424778940z thank you for the report and for the issue. 2.9.0 will be released next week.

zmrubin, 2017-11-13:
Looks like things were moved around, and this fix was undone. Where does teensy.py / teensy.py's functionality live now?

ivankravets, 2017-11-16:
@zmrubin See https://github.com/platformio/platform-teensy/blob/develop/builder/main.py

scott620, 2018-12-04:
@ivankravets 
I am having this exact issue uploading to my Teensy 3.5. Is there a different fix for this board?
I am getting "Teensy did not respond to a USB-based request to enter program mode." However the auto reboot does work when I try flashing with the Arduino IDE. Its strange, because my very first program uploaded no problem at all. Any help is appreciated! Thank you.

PaulStoffregen, 2018-12-04:
Auto-reboot requires software to work correctly on both sides of the USB cable.  Every Teensy has a button dedicated for entering bootloader mode, because plenty can go wrong on either side.

To get an idea which side might be the problem, maybe try programming your Teensy using PlatformIO, so it has no code running that came from Arduino.

Then quit PlatformIO and run Arduino again.  Does auto-reboot work from Arduino on the first try, while the Teensy still has code that was created by PlatformIO?  If that works, it's a good sign PlatformIO is putting good code onto the Teensy which properly listens for the auto-reboot request from your PC.

Likewise, after Arduino has programmed the Teensy with known-good code, try running PlatformIO.  This can serve as a double confirmation whether the problem is on the PC side not sending the auto-reboot request properly, or on the Teensy side not hearing it and entering bootloader mode.

commit 6356e49f (2016-04-09): Automatically reboot Teensy board after upload when Teensy Loader GUI is used // Resolve #609