platformio/platformio-core #610: Globally declared enum yields a series of unexpected compilation failures

dorkusprime, 2016-04-09:
What kind of issue is this?

- [X] Bug report. If you’ve found a bug, please provide information below.

### Configuration

**Operating system**:
OS X El Capitan 10.11.3 (15D21)

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.8.6

### Description of problem

#### Steps to Reproduce

1. Implement `platform_test.ino` code as-included
2. `platformio run`

### Actual Results
```
[Sat Apr  9 12:36:22 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -DPLATFORMIO=020806 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant src/tmp_ino_to.cpp
src/tmp_ino_to.cpp:3:17: error: variable or field 'myFunction' declared void
void myFunction(MyEnum bacon);
^
src/tmp_ino_to.cpp:3:17: error: 'MyEnum' was not declared in this scope
scons: *** [.pioenvs/uno/src/tmp_ino_to.o] Error 1
===================================================== [ ERROR ] Took 0.41 seconds =====================================================
09Apr2016 12:36:23 <wild@WildBookPro:~/Documents/Arduino/platformio_test> platformio run
[Sat Apr  9 12:36:37 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -DPLATFORMIO=020806 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant src/tmp_ino_to.cpp
src/tmp_ino_to.cpp:7:17: error: variable or field 'myFunction' declared void
void myFunction(MyEnum bacon);
^
src/tmp_ino_to.cpp:7:17: error: 'MyEnum' was not declared in this scope
platformio_test.ino: In function 'void loop()':
platformio_test.ino:11:19: error: 'myFunction' was not declared in this scope
scons: *** [.pioenvs/uno/src/tmp_ino_to.o] Error 1
===================================================== [ ERROR ] Took 0.39 seconds =====================================================
```

### Expected Results
Successful compilation

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
```

**Source file (`platform_test.ino`) to reproduce issue:**
```ino
enum MyEnum { X, Y, Z };

void setup()
{
  Serial.begin(9600);
}

void loop()
{
  MyEnum crunchy = X;
  myFunction(crunchy);
}

void myFunction(MyEnum bacon)
{
 if (bacon == X) {
   Serial.println("x");
 } else {
   Serial.println("not x");
 }
}
```

### Additional info
- This code compiles successfully in the Arduino IDE
- Several other Arduino sketches have successfully compile on this same setup

ivankravets, 2016-04-09:
Temporary solution:
1. Rename file to `.cpp`
2. Place on the top of file `#include <Arduino.h>`

dorkusprime, 2016-04-09:
Confirmed.  The following code compiles successfully.

```cpp
#include <Arduino.h>

enum MyEnum { X, Y, Z };
void myFunction(MyEnum bacon);

void setup()
{
  Serial.begin(9600);
}

void loop()
{
  MyEnum crunchy = X;
  myFunction(crunchy);
}

void myFunction(MyEnum bacon)
{
 if (bacon == X) {
   Serial.println("x");
 } else {
   Serial.println("not x");
 }
}
```

commit 00b76e6f (2016-04-26): Refactor source code converter from "*.ino" to "*.cpp" // Resolve #610
closed 2016-04-26 by ivankravets