platformio/platformio-core #611: Arduino framework for STM32

ivankravets, 2016-04-09:
- @rogerclarkmelbourne https://github.com/rogerclarkmelbourne/Arduino_STM32/
- http://www.stm32duino.com/viewtopic.php?f=41&t=728

ubis, 2016-06-09:
I added support for STM32F103Rx: https://github.com/ubis/STM32duino-Framework
I could add more devices and flashing modes and then request PR.

astrohome, 2016-07-06:
**ubis**'s library does not work neither under Linux nor under OS X.

albertoferna, 2016-07-25:
I'm just testing platformio and I don't know much about its inner workings. However, I have tried to use ubis repo with Linux and it works under some conditions. There seems to be an issue with spaces in the directory the project is in. I followed the same procedure, if I do not use spaces the build works, if I use spaces it fails.

madias123, 2016-09-06:
Are there any news on STM32duino and PlatformIO?

madias123, 2016-09-06:
ok, doesn't work anymore with IDE 1.4.0 | CLI 3.0.0b13
even with Command line initialization (I use it on windooze)

File "C:\Users\madias\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\madias\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "C:\Users\madias\.platformio\platforms\stm32duino\builder\frameworks\arduino.py", line 31:
platform = env.DevPlatform()

ivankravets, 2016-09-10:
@madias123  we have some progress on this issue. See https://github.com/platformio/platform-ststm32/pull/4

Big THANK YOU for @ubis

ivankravets, 2016-09-10:
Resolved in https://github.com/platformio/platform-ststm32/pull/4#issuecomment-246117685

closed 2016-09-10 by ivankravets